motioneye-project/motioneyeos #1033: Use Linux network share for file storage

Kazz3r24, 2017-07-12:
I just finished setting up my camera with my raspberry pi 3. Works amazingly, only issue I'm finding is the ability to upload footage to my linux server which is accessible through the network. I'm noticing that in the motioneye.log file it's strictly attempting to connect to an smb share, is it possible at all to have this configured to upload to a linux file system instead?

Thanks!

ccrisan, 2017-07-12:
I'm sorry but I fail to understand the issue here. Let's take the problems step by step.

1. Did you set up motionEyeOS to write your media files to a SMB share (aka NAS)?
2. Do you get any errors? What is the exact problem that you encounter? Log files always help.
3. What do you mean by uploading to a linux filesystem? You can attach an external disk that's formatted as e.g. ext4. Or you could enable uploading to external services (cloud or ftp).

ccrisan, 2017-07-13:
Your network share configuration looks invalid. There shouldn't be any "D:\" or anything of that sort in a network share:

    access to network share //192.168.0.221/D:\CCTV failed: cannot mount network share

See [this](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#network-shares) troubleshooting article for details.

FTP upload support is quite recent. Versions 20170611 and 20170710 have this option.

closed 2017-07-13 by ccrisan

Kazz3r24, 2017-07-13:
I have resolved the issue on my end. At least I believe I did. Got a successful connection message.
Going back to your response, removing the D: drive from the share name didn't change the result, it still couldn't mount it.

I was able to successfully do a test on my linux server using:
Share Name: cctv/Camera1
<using my login creds for linux server>
Root Directory: /

Then when I proceed to test I get Accessing network share succeeded.

Thank you.