motioneye-project/motioneyeos #1034: RPi Zero 1 frame per 2 minutes

aolszak, 2017-07-14:
Hi,

Seems like I'm having really low framerate.

I'm using RPi Zero and IP Camera with resolution 1280x720. I've configured it as Network Camera and take the RTSP stream from the IP Camera.

Desktop software gets 25fps with that IP camera, but RPi gets only 1 frame per 2 minutes (or longer).

Tweaking the settings did not worked.

My other RPi Zero W works not bad with CSI camera (around 5fps), so I think there is no installation problem.

What I've found is that the camera gets a new frame every time I'm updating the settings - what is already faster than normal work frame rates.

'top' is showing load in range 1.5 - 2.1

Any ideas what could be the problem? 

BTW Thanks for great software ;-)

ccrisan, 2017-07-14:
Have you tried disabling motion detection? Did you tweak the frame rate dimmer preference in the settings panel?

aolszak, 2017-07-14:
The only thing which is selected as ON is Video Device - so Motion Detection is off all the time.

I tried the frame rate dimmer - not helping.

I tried to set lowest settings but same low frame rate.

Seems like there is some kind of other problem, but it's hard to spot it.

ccrisan, 2017-07-14:
Given the relatively big sysload you have, identifying the process that eats most of the CPU would be helpful. It probably is either motion or python/motionEye.

aolszak, 2017-07-14:
It's motion with 90-95%.

ccrisan, 2017-07-14:
Then you should probably talk to motion guys. The CPU usage could also be legit, given your stream resolution, though. In any case, there's nothing motionEye can do about this.

closed 2017-07-14 by ccrisan