motioneye-project/motioneyeos #1036: Set static IP on existing installation

synaesthesiac, 2017-07-17:
On an existing installation using the provided image, is it possible to change the LAN ip address to a static address? Would rather avoid reimaging as it's difficult to get physical access. Reason: WiFi used to access the device, LAN used to directly connect to the camera.

jasaw, 2017-07-17:
Assign static IP to your MAC address at your DHCP server?

synaesthesiac, 2017-07-18:
When I said LAN used directly connected to the camera, I mean pi > LAN cable > camera. There is no DHCP server.

ccrisan, 2017-07-18:
What's wrong with setting a static IP from the settings panel, in the *Network* section?

synaesthesiac, 2017-07-18:
Because that is only relevant for the wireless which is used for the management. I see nothing there for the LAN when WiFi is in use.

ccrisan, 2017-07-19:
Unfortunately motionEyeOS doesn't work that way. It expects only one network connection. If you happen to also plug in the ethernet cable, it will start a DHCP server on that interface but the static IP address is used only once, for the first network interface it detects (wlan0, eth0, in this order). If you need fancier network setups, please use motionEye on top of a general purpose Linux distro.

closed 2017-07-19 by ccrisan