motioneye-project/motioneyeos #1040: (former detected) camera not detected anamore

merlinuwe, 2017-07-19:
Hi there,

I just installed motioneyeos again, because my camera isn't (but was one day before) detected. Here is the output of the relevant commands:

The camera is this: http://www.waveshare.com/rpi-camera-h.htm
bought here:
https://www.sertronics-shop.de/Raspberry-PiCo/Raspberry-Pi/Kameras/NoIR-Kamera-fuer-Raspberry-Pi-mit-Fisheye-Lens-einstellbarem-Fokus-und-Infrarot-LEDs/#lightbox[warenkorbgallerie]/4/

The two IR-LEDs are light up both (red). 

What can I do to get the camera work again?

`
[root@motioneyeos dev]# **lsmod**
Module                  Size  Used by    Not tainted
ipv6                  346619  0 [permanent]
brcmfmac              186310  0
brcmutil                5661  1 brcmfmac
cfg80211              430868  1 brcmfmac
snd_soc_bcm2835_i2s     6290  0
rfkill                 16134  2 cfg80211
snd_soc_core          126350  1 snd_soc_bcm2835_i2s
snd_bcm2835            20395  0
snd_pcm_dmaengine       3391  1 snd_soc_core
bcm2835_gpiomem         2972  0
snd_pcm                75620  3 snd_soc_core,snd_bcm2835,snd_pcm_dmaengine
snd_timer              19476  1 snd_pcm
snd                    52522  4 snd_soc_core,snd_bcm2835,snd_pcm,snd_timer
spi_bcm2835             6690  0
uio_pdrv_genirq         3164  0
uio                     8192  1 uio_pdrv_genirq
i2c_dev                 6051  0
i2c_bcm2708             4834  0
bcm2835_wdt             3225  1
bcm2835_v4l2           38417  0
videobuf2_vmalloc       5187  1 bcm2835_v4l2
videobuf2_memops        1332  1 videobuf2_vmalloc
videobuf2_v4l2         16471  1 bcm2835_v4l2
videobuf2_core         20996  2 bcm2835_v4l2,videobuf2_v4l2
v4l2_common             3865  2 bcm2835_v4l2,videobuf2_v4l2
videodev              133090  4 bcm2835_v4l2,videobuf2_v4l2,videobuf2_core,v4l2_common
media                  11573  1 videodev

[root@motioneyeos dev]# **ls /dev/video***
ls: /dev/video*: No such file or directory
[root@motioneyeos dev]# v4l2-ctl --list-devices
Failed to open /dev/video0: No such file or directory

[root@motioneyeos dev]# **dmesg | grep v4l**

[root@motioneyeos dev]# **modprobe bcm2835-v4l2**

`

ccrisan, 2017-07-19:
Does it work work with Raspbian?

merlinuwe, 2017-07-19:
It has worked a day before with motioneyeos, but I will test it soon and give you feedback.

(BTW: A usb-camera (Logitec hd720p) is detected without problems with motioneyeos.)

ccrisan, 2017-07-19:
In this cases I strongly suspect hardware problems. You may want to check out your ribbon cable connection as well.

merlinuwe, 2017-07-19:
Yes, I tested the connection by changing the long to the original (short) cable. The same result.

On an other raspian, (after enabeling the camera-interface with raspi-config) I tested with this command:

`
pi@raspberrypi[/dev]{70}: $ **raspistill -o test.jpg**
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates
`
I think, I have tested all things that could be possible, now.

I spoke with the seller, I can send the camera back to him. 

Thank you for your help!

closed 2017-07-19 by ccrisan