motioneye-project/motioneyeos #1042: Port forwarding loses title bar and footer bar overlays

websitefreedom, 2017-07-19:
I'm using motioneyesos with motion settings on RPi zero wireless and love it - thanks! :-)

Accessing via my LAN works fine. I get the title and footer bar overlays with buttons etc...
[image]
Setting up port forwarding, I log in and find I get the video stream but without title or footer bar overlays so I can't view stored pictures or movies...
[image]

Is there a way of seeing the title and footer overlays via port forwarding?

ccrisan, 2017-07-19:
Forward port 80 instead of 8081 or whatever you forwarded.

closed 2017-07-19 by ccrisan

websitefreedom, 2017-07-19:
Okay thanks :-)