motioneye-project/motioneyeos #1044: MotionEye on RPI 3 lost camera after upgrade

JohnDBowne, 2017-07-24:
Hi,

I set up a new location for my MotionEye setup and dusted off the hardware that I last used on Nov 23, 2016. I was using the current firmware at the time. I got the unit setup and running and things were good. Video was working, some small issues with a connection to a samba mount elsewhere. Its got a good power supply (2+ Amps available), and a wired network connection.

I then pressed the Check button which said that I was out of date, I had it install the new firmware. Nothing moved, changed during this time. After it rebooted, it said that I had no camera. 

Looking at /dev, there was no /dev/video? device.  I can however, take stills with raspistill.

The log files don't seem to point at anything happening that looks out of place, with the exception of /var/motion.log, that is older than the other logs, and was written about the time of the firmware update.

```
[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[0:motion] [WRN] [ALL] thread config option deprecated use camera
[0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
[0:motion] [NTC] [ALL] motion_startup: Motion 4.0.1+git37b3595 Started
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
ccuurrll::  ((5566))  RReeccvv  ffaaiilluurree::  CCoonnnneeccttiioonn  rreesseett  bbyy  ppeeeerr
```
The last couple of lines are odd.

I've tried deleting the existing camera configuration (from the GUI), but when trying to add a camera from the dropdown box, it says that there are no local cameras.

Running v4l2-ctl --list-devices says "Failed to open /dev/video0"
The contents of /etc/modules is: 
```
bcm2835-v4l2 max_video_width=2592 max_video_height=1944
bcm2835-wdt
```
Running vcgencmd get_camera returns "supported=1 detected=1"

The only other thing that is odd is that the output of lsmod is empty. Just the header line and nothing else.

I've rebooting and power cycles, but nothing has been effective.

I'd sure like to get this running, and it was running prior to the update and zero config changes by me.

Thanks in advance,

    johnbo

ccrisan, 2017-07-24:
What happens if you run `modprobe bcm2835-v4l2`?

JohnDBowne, 2017-07-25:
Hi, 

Sorry about the delay, had to work today.

Running modprobe bcm2835-v4l2 produced:

```
[root@camera_01 ~]# modprobe bcm2835-v4l2
modprobe: can't change directory to '4.4.50-v7': No such file or directory
```

```
[root@camera_01 ~]# uname -a
[Linux camera_01 4.4.50-v7 #1 SMP Mon Jul 10 02:50:52 EEST 2017 armv7l GNU/Linux]
```

Looking at /lib/modules:
```
[root@camera_01 etc]# cd /lib/modules
[root@camera_01 modules]# ls
4.4.30-v7/
[root@camera_01 modules]# ls 4*
kernel/              modules.alias.bin    modules.builtin.bin  modules.dep.bin      modules.order        modules.symbols
modules.alias        modules.builtin      modules.dep          modules.devname      modules.softdep      modules.symbols.bin

```
I hope that this is enough information.

Thanks,

  johnbo

ccrisan, 2017-07-25:
This is very odd. Somehow you ended up with a partial upgrade. You have the boot partition of a newer version and the root partition of an older one. This could happen if you upgrade from a very old version when the upgrading mechanism was a bit different.

At this point I recommend using the backup function to backup your settings and restore them over a freshly written OS image. Otherwise you could manually write the data partition of your SD card with the latest version, but that may prove to be more difficult.

closed 2017-07-25 by ccrisan

DirkHeinke, 2017-07-31:
Maybe add a check if the version is to old? I'm now also with a broken system, because I did an upgrade and now I can't longer mount samba shares
```
> modprobe cifs
modprobe: can't change directory to '4.4.50-v7': No such file or directory
```

derekthatcher, 2017-08-01:
I have the same problem as JohnDBowne  - motioneyeos version was 20170212 and I attempted to update (gui) - it reported success but the version number has not changed and the camera does not work anymore. Raspistill works, no /dev/video. I have re-run upgrade several times with the same result. 
Can I run upgrade via ssh? (getting sd card involves a ladder...)

also 4.4.38-v7 directory contents is slightly different.:
# ls /lib/modules/4*
kernel/              modules.alias.bin    modules.builtin.bin  modules.dep.bin      modules.order        modules.symbols
modules.alias        modules.builtin      modules.dep          modules.devname      modules.softdep      modules.symbols.bin

ccrisan, 2017-08-01:
The update mechanism has been considerably changed since 20170212 and things probably went south during the upgrade. The only option you have here is to rewrite the OS from scratch while backing up and restoring the settings.