motioneye-project/motioneyeos #1046: Using curl in a monitor file

gharrigan, 2017-07-25:
I'm attempting to use the following statement in monitor_1 file

GPIO=17
val=$(cat /sys/class/gpio/gpio$GPIO/value)
if [ "$val" == "1" ]; then
 curl -m 5 -o "/data/output/Camera1/test.jpg" -g "http://192.168.1.7/picture/1/current"
fi

When I run the script in the monitor_1 file I get the following error:
mjpg client timed out receiving data for camera 1 on port 8081

I've tested the curl statement in a different script and it gives me the results I wanted.
I've tested other statements in monitor_1 and that works.
I've tried http://xx.xx.xx.xx:80 but the curl statement still tries to use port 8081

I'm using a RPi Zero W 
motionEye Version | 0.36.1
Motion Version | 4.0.1+git37b3595
OS Version | motionEyeOS 20170710

Any assistance would be greatly appreciated.
Thank you in advance.

ccrisan, 2017-07-27:
Please post the output of the following commands:

    cat /data/etc/monitor_1

and

    /data/etc/monitor_1

gharrigan, 2017-07-27:
cat /data/etc/monitor_1

#!/bin/bash
GPIO=17
 
test -e /sys/class/gpio/gpio$GPIO || 
  (echo $GPIO > /sys/class/gpio/export \
  && echo in > /sys/class/gpio/gpio$GPIO/direction)
 
val=$(cat /sys/class/gpio/gpio$GPIO/value)

if [ "$val" == "1" ]; then
 GPIO_text="HIGH"
 curl -m 5 -o "/data/output/Camera1/doorbell.jpg" --get "http://localhost/picture/1/current"
else
 GPIO_text="LOW"
fi
echo "$GPIO:$GPIO_text"
echo 1 1>&2

------------------------
/data/etc/monitor_1

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:05 --:--:--     0
curl: (28) Operation timed out after 5001 milliseconds with 0 bytes received
17:HIGH
1

ccrisan, 2017-07-27:
Your script looks fine. Where did you see the error saying `mjpg client timed out receiving data for camera 1 on port 8081`?

If you saw that in motioneye.log, that is unrelated to your curl request. It's in fact the internal request between motionEye and motion. Otherwise do you see any performance issues when you visualize the stream in your browser?

gharrigan, 2017-07-27:
In fact the error is in the motioneye.log file.
Otherwise no other issues that I can see.

ccrisan, 2017-07-27:
How about increasing the timeout from 5 to something bigger? Just to see if it eventually worls.

gharrigan, 2017-07-27:
I increased the time up to 20 seconds and still doesn't copy the picture over.

I did get some strange behavior when running monitor_1 from CL.
If I quickly hit enter for the command /data/etc/monitor_1 then the picture would copy over.
It didn't seem to work every time but did copy it 4 out of 10 times I tried that.

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 68716  100 68716    0     0    97k      0 --:--:-- --:--:-- --:--:--   99k
17:HIGH
1

jasaw, 2019-07-12:
Closing stale issue.

closed 2019-07-12 by jasaw