motioneye-project/motioneyeos #1047: Changed to static IP and now no video

timfeld, 2017-07-25:
Hello - I have two identical Raspberry Pis with Sony on-board cameras running MotionEyeOS. Both had been working fine until I changed them from DHCP to static addresses. They both worked for a few hours following a reboot but then video output ceased. I tried reverting to DHCP but video still fails. It also fails to mount the SMB share I use to write video files to. Didn’t have a problem with that before and I don’t know if it is related to the video output problem. I have attached four log files (motion, motioneye, messages, and boot).

The log files mention a couple of IP addresses. 192.168.1.9 is the satellite router for my Orbi mesh router. I don’t know why the Pi is trying to access it. 192.168.1.70 is the Windows 10 PC containing the SMB share that stores the Pi’s video files.

I have tried a number of things to try and understand what’s going on. I changed the SMB share to a different PC. I changed to local storage. When I change from SMB share storage to local storage, the video output screen changes from a camera icon with a slash through it to a grey screen that displays the name of the camera, the date/time, and a message in the upper left that says “Unable to open video device.” I tried restoring a backup configuration but it said the restore failed.

I’m guessing since both of them failed the same way after I changed both from DHCP to static, I must have done something wrong (and it’s probably dumb). But I cannot figure out what I did nor how to fix it. Please help. 
[motioneye.log-13.txt](https://github.com/ccrisan/motioneyeos/files/1174197/motioneye.log-13.txt)
[boot.log-3.txt](https://github.com/ccrisan/motioneyeos/files/1174198/boot.log-3.txt)
[messages.log-5.txt](https://github.com/ccrisan/motioneyeos/files/1174200/messages.log-5.txt)
[motion.log-4.txt](https://github.com/ccrisan/motioneyeos/files/1174199/motion.log-4.txt)

timfeld, 2017-08-01:
Any ideas? 
Would it be easier to simply wipe the Pi and start over?

timfeld, 2017-08-08:
Hello? Anyone out there?

jasaw, 2017-08-08:
The log files you've attached are huge, so let me try to understand what happened, and maybe @ccrisan can give some feedback too.

You upgraded the motioneyeos firmware on 22 Jun 2017 after being turned off for 1 month, and since then you've been getting this:

Failed connection to remote camera at 192.168.1.98

> 2017-07-22 19:08:47: [motioneye]    ERROR: failed to get config for remote camera 1 on http://192.168.1.98:81/config/1: Failed to connect to 192.168.1.98 port 81: No route to host
> 2017-07-22 19:08:47: [motioneye]    ERROR: failed to get config for remote camera 1 on http://192.168.1.98/config/1: Failed to connect to 192.168.1.98 port 80: No route to host

and SMB mount of 192.168.1.70 failing

> 2017-07-22 19:09:28: [motioneye]    ERROR: failed to mount smb share "//192.168.1.70/PiCam_Video_Files/FrontDoorCam" at "/data/media/motioneye_192_168_1_70_picam_video_files_frontdoorcam_timf"
> 2017-07-22 19:09:29: [motioneye]  WARNING: Connect error on fd 19: ECONNREFUSED

and every now and then, your windows machine logs in:

> 2017-07-22 19:08:39: [motioneye]  WARNING: 403 GET /login/?_=1500772118986 (192.168.1.70) 1.84ms

You changed the IP address of the motioneyeos device from DHCP (assigned 192.168.1.37) to static (192.168.1.231), and as far as I can tell, there's no problem with going DHCP to static IP.

I don't see the motioneyeos device attempting to access 192.168.1.9 either.

That's the gist of it.

Looks like a networking problem to me.

ccrisan, 2017-08-08:
This is clearly a network connectivity issue. You appear to have set a remote motionEye camera, which is no longer reachable. This is definitely not a motionEyeOS issue.

timfeld, 2017-08-08:
Thank you both! That is very helpful information...I appreciate it.

Tim

closed 2017-08-08 by timfeld