motioneye-project/motioneyeos #1049: upgrade MotioneyeOS

marcorighi, 2017-07-26:
Hi, before to install MotioneyeOS I would like to know if is it possible to use commands similat to "apt update" and "apt dist-upgrade" to upgrade the distribution. Thank you very much for your work.

ccrisan, 2017-07-26:
https://github.com/ccrisan/motioneyeos/wiki/FAQ#i-cant-seem-to-find-apt-get-or-any-other-package-manager-what-gives

closed 2017-07-26 by ccrisan

marcorighi, 2017-07-26:
Drear Calin,
it is not the answer to my question. For example, how can I upgrade motion or motioneye in MotioneyeOS?

ccrisan, 2017-07-26:
That was my point: you can't. The only way to upgrade anything in motionEyeOS is to upgrade the OS itself. Which is why I made it very easy to perform the upgrade and why I do my best to keep the OS image is so small.