motioneye-project/motioneyeos #1050: 256GB USB Mass Storage Device keeps changing path

TamirSlo, 2017-07-26:
Hello,
I am quite new to motion and all of it.. I am trying to output videos and images & the settings are OK for me... the only issue ive got is that somehow every so often, the USB is changing path, e.g. from **/data/media/sda5** to **/data/media/sdb5** ... Its also continuing to b->c; c->d; etc..

Any idea why / ideas of how to make its path permanent?

_Just for the record, I am not disconnecting any USBs. Currently I have 2 USBs( 1 for camera, and one for storage) and I am using Raspberry Pi 3. The camera does not look like its changing paths._

Thanks,
Tamir

jasaw, 2017-07-26:
Can you please do `dmesg` and look for USB related logs?
Without looking at logs, the first thing I would look for is bad USB connection, either from bad USB cable or unstable power supply.

jasaw, 2017-10-11:
I'm not sure how USB mass storage devices are mounted currently, whether they are automounted or not. @ccrisan Can we mount the mass storage device on a directory with its UUID? We will end up with the same mount point each time.

ccrisan, 2017-10-11:
[61-usb-storage.rules](https://github.com/ccrisan/thingos/blob/master/board/common/overlay/etc/udev/rules.d/61-usb-storage.rules) is the file responsible for automounting disks. Currently we create the mount point based on the device name (e.g. `sdb1`).

Having the mount point as UUID is a good idea - too bad I didn't go with it in the beginning. Now it would break a lot of existing setups.

Anyways, OP's problem here may be caused by some other deeper problems. The system would normally allocate sdb instead of sda only if sda would somehow not be released. For some reason the OS still thinks some sda is attached. It may be a power issue with the USB disk though.

TamirSlo, 2017-10-11:
One of my ideas to overcome this was to use the 256GB usb as an external hard drive that the raspberry pi is booting to, but as its 256GBs it would take very long time to format and write the image for the whole disk..

I’d live to hear any other advices or ideas 👍🏻

chris7148, 2018-08-07:
Hi @ccrisan, thanks for a superb piece of software! I have a similar issue - and appreciate changing the default now would break a lot of people's setups. Would it be possible/practical without messing up your code to expose an advanced config option to allow users to override default behavior to use UUID mounting points instead? I would be very happy to assist in testing, etc!

jasaw, 2019-07-12:
@ccrisan It would still be nice to support UUID mount point. I'm thinking maybe switch the mount point to UUID type, and have /dev/sd* as a symlink to the UUID mount point. Thoughts?

ccrisan, 2019-07-27:
I'm not against using UUIDs but at this point I think we've got other priorities to deal with. Switching to UUIDs may sound simple but I believe there will be implications everywhere, including firmware updating.

closed 2019-10-30 by jasaw