motioneye-project/motioneyeos #1051: Got the Raspberry Pi +MotioneyeOS + Port Forward to Work -but security/private pictures

baker195, 2017-07-27:
Thank you for a great program(s) ! Is there a way to keep the camera pictures private when I port forward to the internet? I see the picture framed with motioneyeos... but anyone can see this too..
Suggestions?

Thanks
Scott

Chiny91, 2017-07-27:
Anyone can see your motionEyeOS server ?   That would be scary.    Obviously you have a VERY difficult admin password....  and an equally complex but different username and user password.

Changing the http port will fractionally (not a lot) add to security but security through obscurity is not generally highly rated.

A vpn, perhaps OpenVPN on your router, or a handy Pi, would be better.   Maybe add some firewall rules to prevent any IP camera from talking to the net at all.   Your motionEyeOS box will tend to want (it's the default ISTR) net access for the time, and for checking network access but even there you could point it to LAN servers.

Or, copy (rsync) pictures to a hardened box and vpn into that.   Or, motionEyeOS does provide facilities to upload media files to various services (Google, er, Dropbox ISTR) and look there.

There are probably millions of R Pi boxes exposed to the net and it is a tribute to their design that they appear to survive unscathed.

baker195, 2017-07-27:
Brilliant ideas 
Will look into local network with upload to Dropbox 
Thanks !!

Baker195@aol.com

baker195, 2017-07-28:
Thanks again.

closed 2017-07-28 by baker195