motioneye-project/motioneyeos #1053: Pi Zero - Microsoft Lifecam Errors

pypipy26, 2017-07-27:
From motion.log:

[1:ml1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
[1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
[1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1:ml1] [ERR] [VID] v4l2_set_mmap: Error requesting buffers 4 for memory map. VIDIOC_REQBUFS: Device or resource busy
[1:ml1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device

Any ideas how to resolve this? The camera works with my Lubuntu laptop, not sure what's going on with motioneyesos.

Thank you!

jasaw, 2017-07-31:
I don't know much about v4l2 drivers. Try the below steps first?
- make sure your camera is supported. [Here's](http://elinux.org/RPi_USB_Webcams) a list of supported USB webcams.
- try your camera on Raspbian, see if you get the same problem. This should tell us whether it's a video driver issue in motioneyeos.

jasaw, 2017-10-11:
@pypipy26 Have you resolved this issue?

jasaw, 2019-07-12:
Closing issue due to lack of follow up from issue author.

closed 2019-07-12 by jasaw