motioneye-project/motioneyeos #1056: IP Camera Issue

pascone10, 2017-08-01:
I have an IP Camera EziView Model: DS-2CD1141-1. When I attempt to add it to MotionEyeOS it says not a supported network camera. The camera does support RTSP on Port 554. Please help me determine what cameras I can use or if there is something I need to adjust in the camera software. I appreciate any help in advance.

pascone10, 2017-08-01:
OK I found my issue. The IP address is similar to HikVision cameras. The URL is RTSP://ipaddress/    
I was using HTTP://ipaddress. 
I hope someone else finds this useful. Is there a manual or doc on the OS i can reference for all the settings?

Also I would like to thank the creator of this software!

pascone10, 2017-08-01:
Im not sure if I should create a new issue. My question is I am using a WD passport 2TB hard drive to record video. It is on my network via my Linksys Router USB port. The video is very spotty and when there is motion the person in the video and images appear and disappear. I have changed all settings I could find from resolution, frames (in both camera and MotionEyeOS. Nothing seems to fix this. Even the live streaming is having problems. 

Any Ideas or advice?

ccrisan, 2017-08-02:
> OK I found my issue. The IP address is similar to HikVision cameras. The URL is RTSP://ipaddress/    
I was using HTTP://ipaddress. 
I hope someone else finds this useful. Is there a manual or doc on the OS i can reference for all the settings?

Good to know. Unfortunately there's no reference for all settings. The closest we have to that is the [configuration options](http://htmlpreview.github.io/?https://github.com/Motion-Project/motion/blob/master/motion_guide.html#Configuration_OptionsTopic) that motion supports. Most of them are covered by friendly UI elements in motionEye.

> Also I would like to thank the creator of this software! 

You're welcomed!

> Im not sure if I should create a new issue. My question is I am using a WD passport 2TB hard drive to record video. It is on my network via my Linksys Router USB port. The video is very spotty and [...]

Yes, you should create a new issue, but given the way it sounds, it may be more related to [motion](http://htmlpreview.github.io/?https://github.com/Motion-Project/motion) than to motionEye(OS). The motion log file would be very useful.

closed 2017-08-02 by ccrisan

jasaw, 2017-08-03:
> The video is very spotty and when there is motion the person in the video and images appear and disappear.

@pascone10 Was the person walking in and out of the camera view? Was it a low frame rate issue? 1 or 2 fps? What do you see when live streaming? What hardware is MotionEyeOS running on?

pascone10, 2017-08-03:
https://www.dropbox.com/sh/4zw63u383ypoo3o/AACb4MXltd2qOdQUgtgeyvf7a?dl=0

Here is a link to the motion eye logs. 

I am using a Rasberry Pi 3. The person was walking in and out of the frame. I enabled frame detection so I could see the red box where the camera was picking up the person but they were not there. I know the path taken and I could see the red box was following the person but they did not appear in the frame. 

OR at times the would be there and the video would lag and show there were 3 to 5 ppl ( same person walking). I tried multiple different frame rates. The bottom of the camera on main view shows 4.4/1.2fps. 

I have the frame rate under the video device tab set for 15fps. 

Do the camera settings inside the camera need to match the Motion software for things like FPS or resolution?