motioneye-project/motioneyeos #1058: Wears out SD card continuously writing log messages

romanrm, 2017-08-03:
Hello,

Logging in to my installation, I notice that it logs the following message to dmesg many times a second:

```
[Thu Aug  3 10:28:57 2017] uvcvideo: Failed to query (SET_CUR) UVC control 2 on unit 2: -110 (exp. 2).
[Thu Aug  3 10:28:58 2017] uvcvideo: Failed to query (SET_CUR) UVC control 2 on unit 2: -110 (exp. 2).
[Thu Aug  3 10:28:58 2017] uvcvideo: Failed to query (SET_CUR) UVC control 2 on unit 2: -110 (exp. 2).
[Thu Aug  3 10:28:58 2017] uvcvideo: Failed to query (SET_CUR) UVC control 2 on unit 2: -110 (exp. 2).
[Thu Aug  3 10:28:58 2017] uvcvideo: Failed to query (SET_CUR) UVC control 2 on unit 2: -110 (exp. 2).
[Thu Aug  3 10:28:59 2017] uvcvideo: Failed to query (SET_CUR) UVC control 2 on unit 2: -110 (exp. 2).
[Thu Aug  3 10:28:59 2017] uvcvideo: Failed to query (SET_CUR) UVC control 2 on unit 2: -110 (exp. 2).
[Thu Aug  3 10:28:59 2017] uvcvideo: Failed to query (SET_CUR) UVC control 2 on unit 2: -110 (exp. 2).
[Thu Aug  3 10:29:00 2017] uvcvideo: Failed to query (SET_CUR) UVC control 2 on unit 2: -110 (exp. 2).
[Thu Aug  3 10:29:00 2017] uvcvideo: Failed to query (SET_CUR) UVC control 2 on unit 2: -110 (exp. 2).
[Thu Aug  3 10:29:00 2017] uvcvideo: Failed to query (SET_CUR) UVC control 2 on unit 2: -110 (exp. 2).
[Thu Aug  3 10:29:01 2017] uvcvideo: Failed to query (SET_CUR) UVC control 2 on unit 2: -110 (exp. 2).
[Thu Aug  3 10:29:02 2017] uvcvideo: Failed to query (SET_CUR) UVC control 2 on unit 2: -110 (exp. 2).
[Thu Aug  3 10:29:02 2017] uvcvideo: Failed to query (SET_CUR) UVC control 2 on unit 2: -110 (exp. 2).
[Thu Aug  3 10:29:02 2017] uvcvideo: Failed to query (SET_CUR) UVC control 2 on unit 2: -110 (exp. 2).
[Thu Aug  3 10:29:02 2017] uvcvideo: Failed to query (SET_CUR) UVC control 2 on unit 2: -110 (exp. 2).
[Thu Aug  3 10:29:03 2017] uvcvideo: Failed to query (SET_CUR) UVC control 2 on unit 2: -110 (exp. 2).
[Thu Aug  3 10:29:03 2017] uvcvideo: Failed to query (SET_CUR) UVC control 2 on unit 2: -110 (exp. 2).
```
This is also continuously written to /var/log/dmesg.log, and none of the usual Raspberry Pi tricks to move the log directory into RAM are applied. The size of the log file rapidly reached 2 MB already. This means the SD cards will wear out and die at an accelerated rate.

(As for the log message itself, it doesn't seem to cause any problems, the camera works fine and the image is perfectly okay.)

ccrisan, 2017-08-03:
That looks like a problem with the uvcvideo driver and I don't know what to do about that.

romanrm, 2017-08-03:
The bug report is meant to note that all logging is currently directed to disk, which is unnecessary and reduces the flash lifespan. For a system primarily targeted to be used on SD cards, redirecting all logs to RAM should be a must.
http://www.zdnet.com/article/raspberry-pi-extending-the-life-of-the-sd-card/
http://oils-of-life.com/blog/linux/reduce-writes-to-the-raspberry-pi-sd-card/ #3

ccrisan, 2017-08-03:
@romanrm you are right, sometimes it makes sense to have logs saved in ram. But most of the times logs are relatively small in motionEyeOS and it's worth persisting them on the SD card so that you can see what went wrong, if something were to go wrong.

I'll reopen this issue with the idea to add an option to choose whether you want logs to be saved on the SD card or you're fine with them living in RAM.

jasaw, 2017-08-03:
@ccrisan Option to log to RAM or disk would be a nice feature to have.

ccrisan, 2017-08-06:
I have written a [dedicated wiki page for logging](https://github.com/ccrisan/thingos/wiki/Logs#logs-in-ram). Most of the docs will be transferred to thingOS, as that's where they belong. I'll make sure to add references from motionEyeOS so that motionEyeOS users can also find them.

closed 2017-08-06 by ccrisan