motioneye-project/motioneyeos #1059: Every 3 days Motioneyeos goes down and the Pi's needs to be unplugged.

shadowfxd, 2017-08-03:
I have motioneyeos on a pi zero, pi2 and pi3. I've tried fast camera and regular. I tried local motion and a server. No matter what I do, it becomes unresponsive every 3 days and needs to be completely unplugged and plugged back in to work again. I've tried keep motion alive even. I'm running all cams 640x480 @ 7-9FPS, 25% quality. All with different IPs and streaming ports. All have the OC of their name (pi2 has pi2 OC, pi zero has pi zero OC from the menu). All are at 256MB Ram. All have lights off. Every other setting is stock.

ccrisan, 2017-08-04:
You need to dig deeper to find out what's going on. An idea would be to try to recover the log files out of a dead unit and see what's going on. Another idea would be to try a different network connection as this is probably the only common thing between them that may go wron.

shadowfxd, 2017-08-19:
All cameras worked fine until recent releases started making it less and less usable. It's still shutting down every 3 days on 20170710 release.

ccrisan, 2017-08-19:
Please attach your log files here.

shadowfxd, 2017-08-19:
I tried to but it keeps saying "We don't support that file type". Is there a trick to attaching them?

shadowfxd, 2017-08-19:
Here's my [LOGS](https://drive.google.com/drive/folders/0B6p_QTaiYAfZSVRxMmZnZmZmTmc?usp=sharing).

pnauta, 2017-08-20:
There are multiple brownout messages in these logs, clearly indicating your
power supply is not coping with it's task.  Swap out your supply and see
what happens.

shadowfxd, 2017-08-20:
I saw that before i posted and tried 5 different power supplies, all were 5V 2.4Amps. All that's plugged into the pi 2 usb is the wifi dongle, so it shouldn't need more than 2.4amps @ 5V.

ccrisan, 2017-08-20:
It usually is about cables and not the PSU itself. No reasonable PI setup draws more than 1A, at least not in my household. But if the cables are relatively long/thin, the voltage drop on them becomes significant and you go below 4.75V. In these cases, many users choose PSUs rated at 5.25V.

shadowfxd, 2017-08-20:
I just tried a 5.25V with a thick anker micro usb cord and it still says brownouts... The other cams that are having the same problem have no brownout messages though.

ccrisan, 2017-08-20:
Could you temporarily replace your PI (just the board) so that we can rule out a broken board?

shadowfxd, 2017-08-20:
I tried it... still brownout... Do you think it's still brownouts that's the problem?

ccrisan, 2017-08-20:
That's a very plausible explanation. You could try and measure the voltage across pins 2 and 6 and check for yourself to see what's going on with the power.

shadowfxd, 2017-08-20:
I may have found it... it may have been the extension cord i used. I tested a new one, no brown outs on same Pi 2 setup as I originally had problems with. I'll have to see if it's still up in 3 days.

shadowfxd, 2017-08-20:
scratch that, that didn't last long either... back to brownouts.

jasaw, 2017-08-21:
Can you please try your Pi Zero on its own, without camera, without monitor, without keyboard, just let it idle. See if you still get brownout. If you still get brownout, try the exact same PSU and Pi Zero at a different location, office or friend's house. This should tell you whether your mains is too noisy.

shadowfxd, 2017-08-21:
It's a Pi 2 with the brown-outs, but I'm using it on a APC UPS. It should have the cleanest power and using a canakit 2.4A @5v PSU, which I'm also using on 3 other motioneyeos setups. I've been using this setup for a year now, without any problem until lately. I was using a extension (plain without any surge protection) but I took that down and went straight to the APC UPS. This is also happening on another Pi 2 that has zero brown-outs, in another room, with exact memory card, PSU, tplink wifi and APC UPS.

jasaw, 2017-08-21:
Let's see if I understand your situation. 
- You've got multiple of Pi Zero, Pi 2, and Pi 3.
- All your Zero, 2, and 3 are "overclocked" from MotionEye menu.
- Out of your multiple overclocked Pi 2 boards, one of them started showing brownout problem.
- All your Zeros and 3s do not have brownout problem.

Does that summarize your situation?

The reason I asked you to use Zero is because MotionEye overclock setting for Zero is actually the factory default for Zero (1GHz). This rules out brownout caused by overclocking.

shadowfxd, 2017-08-21:
I use 2 pi 2, 1 pi zero and 1 pi 3 for motioneyeos. both pi 2's have been giving me problems where they need to be reset every 3 days or they don't show up via browser or ip cam apps (on windows, linux or android). One of the two show a brown-out problem, but has the same problem as the other pi 2.

jasaw, 2017-08-21:
I would set the clock back to default i.e. no overclock. See if problem persists.

shadowfxd, 2017-08-21:
I actually tried that earlier. I put overclock to "none" still had brown outs.

jasaw, 2017-08-21:
> All cameras worked fine until recent releases started making it less and less usable. It's still shutting down every 3 days on 20170710 release.

Are you able to try the previous release that was stable? Just want to rule out hardware problem.

shadowfxd, 2017-08-22:
I put in a different memory card with motioneyeos 20160705 and zero brownouts and no resets so far.

shadowfxd, 2017-08-22:
I finally figured it out. Everything worked just fine until I added back in the commands to turn off the power and activity LEDs. Then I got the brownout messages again. I then commented them out and brownouts went away again. Can you please fix this? It used to work on old versions, these are the commands I used to add without brownout problems:

mount -o remount,rw /
mount -o remount,rw /boot
nano /boot/config.txt

at the bottom of /boot/config.txt add this:

PI 2:

# Disable the ACT LED.
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off

# Disable the PWR LED. DONT USE!!!
#dtparam=pwr_led_trigger=none
#dtparam=pwr_led_activelow=off

shadowfxd, 2017-08-23:
is there a way to have brownouts skip disabled PWR LED commands? Or a way to completely disable brownouts?

jasaw, 2017-08-23:
Can you please run `vcgencmd get_throttled` periodically? It shows whether under-voltage event has occurred.

```bash
vcgencmd get_throttled
# 0: under-voltage
# 1: arm frequency capped
# 2: currently throttled
# 16: under-voltage has occurred
# 17: arm frequency capped has occurred
# 18: throttling has occurred
```

shadowfxd, 2017-08-23:
throttled=0x50000

with PWR LEDs off or on.

jasaw, 2017-08-23:
throttled=0x50000 means under-voltage and throttling.
Can you this vcgencmd with both PWR LED on and off?

shadowfxd, 2017-08-23:
it's the same with the PWR LED on or off.

jasaw, 2017-08-23:
Is your Pi2 currently overclocked? If it is, try without overclock, see if it changes. It usually points to power issue if you get throttled=0x50000 or throttled=0x50005.

What does vcgencmd give you on motioneyeos 20160705?

shadowfxd, 2017-08-23:
It's not a power issue.  I tried this on several Pi 2's with different power supplies and different memory cards and all have no issues until I turn off the PWR LED. I even tried to install motioneye on raspbian and turned off the PWR LED and no brownouts. Only in motioneyeos do I have this problem.

jasaw, 2017-08-23:
@ccrisan Any thoughts on what might be the issue here?

shadowfxd, 2017-08-23:
I wish I knew, I'm just a novice in this, or I'd have better feedback. I run the Pi 2 through my usb meter and it showed over 5V with LED on or off, yet the message.log still shows brown outs with LED off...

[pics](http://imgur.com/a/LkkRm)

ccrisan, 2017-08-23:
This is interesting. Guys I have to admit I had no idea about the `vcgencmd get_throttled` option.

If the `cmdline.txt` settings cause these problems with the recent version whereas with older ones it works fine, it might just be a problem with the firmware used in the latest version. If this is the case it's probably already fixed in newer firmware versions that will be included with the next motionEyeOS release. By the way, a new prerelease is on its way.

Another thing would be to make sure the instability problem is actually caused by (related to) the false brownouts. @shadowfxd can you (did you) run your setup without the cmdline tweaks long enough to make sure the stability problem is no longer manifesting?

ccrisan, 2017-08-23:
@shadowfxd as far as I remember, the brownout threshold is 4.75V, so you needn't worry about having 5.06V :)

shadowfxd, 2017-08-23:
it's been up for 6 hours now without any brownout. Typically I was getting a brownout every minute with the PWR LED off. The PWR off LED commands work perfectly for Pi 2 in Raspbian without any lightning symbols, so I have no idea why it has brownouts every minute for 3 days, then has to be rebooted by pulling the power cord.

jasaw, 2017-08-24:
@shadowfxd Are you able to find out which GPU firmware you're running on both 20160705 and 20170710 versions?
`vcgencmd version`

You should get something like:
>Jul  3 2017 14:20:31
Copyright (c) 2012 Broadcom
version 4139c62f14cafdb7d918a3eaa0dbd68cf434e0d8 (tainted) (release)

jasaw, 2017-08-24:
I found this [post](https://github.com/raspberrypi/linux/issues/1332#issuecomment-280087412) that talks about turning on/off PWR LED enables/disables throttling on a Pi3. You can disable it by adding `avoid_warnings=2` to /boot/config.txt, but I'm not sure what the implications are.

Don't know if this has anything to do with Pi2, but you're free to try it if you want.

shadowfxd, 2017-08-26:
It has now made it 3 days without a reset, so it was the PWR LED. Now to test the avoid_warning and see if that does anything.

shadowfxd, 2017-08-29:
I've now been testing it with motioneye on raspbian with PWR LED off and no problems or lightning bolts. Is there a log in the raspbian version?

ccrisan, 2017-08-30:
@shadowfxd there's a new prerelease out and it incorporates latest versions of the rpi firmware and kernel. Please try this new version and see if the problem still persists: https://github.com/ccrisan/motioneyeos/releases/tag/20170827.

glyndon, 2017-10-13:
It appears to be just a mistaken assumption made in the S11brownout script in /etc/init.d. Not a firmware or hardware issue.
That script tests the brightness of the LED every 30 seconds. If it's too dim (which is is, if we've set it to be off in config.txt) then the script emits the false-alarm brownout warning. 
That script should be monitoring something more directly connected to the power circuitry, and not the LED, which is user configurable.

ccrisan, 2017-10-13:
Can we reliably detect brownouts then? If yes, what would be the method? Is there a `vcgencmd` command for that?

ccrisan, 2017-10-16:
@jasaw the `vcgencmd get_throttled` command is rather cool. I'll update the script and make sure it uses this command instead of reading the GPIO.

ccrisan, 2017-11-01:
Fixed via https://github.com/ccrisan/motioneyeos/commit/2f3b2774f9fa78e1ed57731c2823e806aa8d682b.

closed 2017-11-01 by ccrisan