motioneye-project/motioneyeos #1060: Unable to limit number of still images

charemer, 2017-08-04:
Using pi zero w writing to samba share,  When motion is detected more stills than desirable are being created/saved: this causes I/O and CPU usage issues.  Is there a mechanism for limiting the number of stills shots being saved?  It appears that every frame is being saved as a still whilst motion is being detected.  Being able to limit the number of frames per minute or specify a percentage of frames to save (e.g. 1 frame in 5 = 20%) would be a useful feature.

ccrisan, 2017-08-05:
This could be achieved to some extent by setting `output_pictures` to `first` or to `best`. Currently motionEye sets it to either `on` or `off`. Additional UI settings and some code would be needed to have this option.

ccrisan, 2017-08-05:
https://github.com/ccrisan/motioneye/commit/252fe4f214140d3d84baa84b6af2ad3507e65d85 adds "Motion-Triggered (One Picture)" capture mode for still images. This should be a good enough solution for your problem.

closed 2017-08-05 by ccrisan

charemer, 2017-08-07:
Thank you,  I'll investigate the one picture mode. :)