motioneye-project/motioneyeos #1061: remote viewer: how to force the layout

atvinnys, 2017-08-08:
the layout options (row/column) depends on the width of the web page.
for some reason, on my vertical screen, I can't seem to find a configuration that would tile 2x2 (I have 4 cameras so far).

is there a way to force  the layout?
thanks
Vincent

ccrisan, 2017-08-08:
At the top of your settings panel, you have two preferences: Layout Columns and Layout Rows.

If you have 4 cameras, set Layout Columns to 2. The Layout Rows setting affects the number of frames that fit the window vertically (are entirely visible) at a certain moment. I'd recommend setting it to 1 but, depending on your resolution, you may want it to be 2 as well.

closed 2017-08-08 by ccrisan

atvinnys, 2017-08-08:
what you describe is all good IF those layout settings show up in the preferences.
however, whether or not they show up seems to be depending on  various thins: width of the browser window from what I can tell..

in my case, with Fit Frame vertically On, only the layout rows shows up.
if I have Fit frame vertically Off, **none** of the layout row/column show up.

thus my question on "forcing" the layout w/o the need of the sliders simply because they don't always show up. 

hope this makes sense.
thanks!

ccrisan, 2017-08-08:
If you're on mobile (small screens), things are a bit different indeed. You won't be able to have more than one frame in a row. It wouldn't make any sense anyway.

atvinnys, 2017-08-08:
this is not mobile...this is on a rasperty PI  with a vertical 1900x1200 vertical screen.
I'll see if I can mimic a higher resolution to get motionEye to think it can have 2 columns.

I just wanted to make sure there was no obvious layout settings I could set to bypass the slider logic.
Cheers

atvinnys, 2017-08-09:
I upgraded to a pi3 and chromium there seems to behave properly (same screen).
Cheers

Frankthedoc, 2018-02-27:
I have this issue with some new Pi3's attached to an HDTV (will check functioning resolution); I'm using chromium-browser in kiosk mode at start up because this is for a monitoring screen. I'll post back if I can identify the issue, but any advice would be welcome!

RHTizzy, 2018-10-26:
Issue has returned on android tablet using Chrome. Had two colums and 1 row, but changed layout to 1 column and 2 rows to check if it was practical. But then the option for 2 columns disappeared and does not reappear. Tablet is most often used in landscape mode so it is more practical to use 2 columns. Highly peculiar that that option would need to be hidden in the GUI for whatever reason