motioneye-project/motioneyeos #1062: Hello! I need your help...

misson112, 2017-08-08:
[image]

On motionEYE OS, I want a use Odroid Xu4 board with multiple camera system.

All camera was recognized but like a image, three camera was not working. Just only one camera was worked.

[image]

What is error capturing first image?

How can I solve this problem?

Note that this system can works in raspberry pie3 with too slow FPS.

jasaw, 2017-10-11:
After reading through motion source code, I can tell you that "Error capturing first image" means motion is unable to get any image from the camera, not a single image.

Those cameras with errors, what are they? network cameras? USB cameras? Could be a driver issue.

jasaw, 2018-06-05:
Closing issue due to lack of follow up.

closed 2018-06-05 by jasaw