motioneye-project/motioneyeos #1063: Setting a Working Schedule appears to prevent motion detection from working

Halle, 2017-08-09:
Greetings,

Thank you for this great project. I have the impression that setting a working schedule prevents any motion detection either inside or outside of the set schedule – I have never had any images saved when I have Working Schedule set to on, in about a week of using it, including at in-schedule times when I've walked in front of the camera. When I don't have the Working Schedule setting on, I always have some motion detected images within what would have been the working schedule (and many hundreds outside of it, which is why I am trying to troubleshoot the use of the working schedule). I am using a Pi Zero W with a Pi NoIr camera (the latest, small camera model). My system version info:

motionEye Version | 0.36.1
Motion Version | 4.0.1+git37b3595
OS Version | motionEyeOS 20170710

I would be happy to troubleshoot this further and provide more info if you could let me know where to focus investigation, thanks again!

Halle, 2017-08-10:
Ok, this appears to be a manifestation of this issue, can be closed (it would be great to get a UI hint about this or wiki entry on it if possible!): https://github.com/ccrisan/motioneyeos/issues/650

jasaw, 2017-08-11:
Added a wiki page about [working schedule](https://github.com/ccrisan/motioneyeos/wiki/Set-up-a-working-schedule).

closed 2017-08-11 by jasaw

geoffp13, 2019-07-04:
Frustration is not the word for it. ccrisan commented somewhere that the schedule for the day cannot overrun that day ie: Tue from 21:00 to 08:00 on Wed is not allowed and is the reason motion detect does not work.
OK so I want to observe motion from Tue 21:00 (9pm) to Wed 08:00 (8am) the next morning.
Of course I can put from 21:00 to 00:00 (midnight) for each day and then put 00:01 (1 minute after midnight) in OOOOH!! where - the next day - of course - but then I cannot put 21:00 to 00:00 for that day ??? 
The strange thing is that after discovering that each camera needs a "thread" (no metion in the manual and not provided in the build) and got that working the schedule for that night from 21:00 to 08:00 worked perfectly - brilliant. So I left well alone thinking all my problems were over !! Not so because the next night nothing was recorded. I did not reboot or change any settings from the first success.
Clearly ccrisan has done a great job so far but has missed out on real world applications.
School children as well as grown ups are going to realise some fore thought is missing here and maybe thats the challenge for them to figure out how to incorporate an overnight detection situation.
It was pointed out to me ealier that the source code is available. I wonder if they have the classroom time to figure it out - I certainly don't !!
ttfn
[image]

jasaw, 2019-07-05:
@geoffp13 We have an existing enhancement request to rework the working schedule. See https://github.com/ccrisan/motioneyeos/issues/70

Problem is, nobody has the time to implement it. Seeing that this is an open source project, you are more than welcomed to implement it and do a pull-request.

geoffp13, 2019-07-05:
Thank you for your response. Thats an interesting idea. Although I am able to write programs and logic schematics I am not sure I am up to the standard required here. I will find the source code and take time to study and see if I can assist.
I must admit that since my Pi 4 and Pi Zero W arrived with various addons I have been having fun with Python and various hardware.
ttfn