motioneye-project/motioneyeos #1064: Notification when Motion detected and when motion ends

developera35, 2017-08-09:
Hio Ccrisan 
Thank you verry much for this masterpiece of software;)

I've set up a RPI3 which calls a web hook when motion has been detected, it works nearly perfect. 

Yet for my aplication i need not just one notification as i want to controll the light. 
So if a person enters the room i dim up the light; aslong as there are movements. 

What I need is acess to the information if there is movement in the room  or a frequent pushnotification that there is stil movement.

My problem at time is this: 
I've implemented a hysteresis of let say 10 minutes. So if someone enters the room the light wil bi ON for 10min if the person is all time in motion and turns off as i don't receive more information that there is still movement in the room.

If this person stays in this room and causes any movements the light should stay turned on. 

Is there a way to check if there is movement in the room ?

Kind Regards
Lopea

ccrisan, 2017-08-10:
We have added support for running a command when the event ends in upstream motionEye. It will be part of the next motionEyeOS release: https://github.com/ccrisan/motioneye/commit/7ecd924b05f0424fd5abc219f7fd17a8cf739e94.

closed 2017-08-10 by ccrisan

developera35, 2017-08-10:
Thanks for answering that fast, 
When will it be released?