motioneye-project/motioneyeos #1065: Saving Video to a hard drive 1 tb usb

jstarne1, 2017-08-09:
Hello, I am just starting out so I am as new as it gets, but I can follow instructions. I am setting up a Pi3 with the pi cam as a continuous surveillance camera and I want to record to a 1tb hard drive plugged into the usb. So far power has not been an issue , but I am getting an error and I dont have permission to write to this volume.   Help a brother out?

jasaw, 2017-08-10:
1. Configure motion to save files in the external hard drive.
   - From the web interface, under File Storage section, you can change the Storage Device and Root Directory. You may need to enable Advanced Settings from web interface.
2. Make sure the path you put in the configuration exists and has the right permissions.

A snippet of the relevant log that shows the error might be helpful.

ccrisan, 2017-08-10:
@jstarne1, like @jasaw said, we need to see some logs. Also, from the top of my mind, you might want to check your filesystem type and make sure you have no write permissions problems on it. EXT partitions would pose such a problem.

jstarne1, 2017-08-10:
Hello all, I reinstalled the image and then did the setup while the hard drive was plugged in and it mounted, after that I was able to select the WD easyshare 1 Tv drive as the recording destination.  Any tips on how large I should make the video segments? I was thinking 8 hours at a time.

ccrisan, 2017-08-10:
So does it work now? Can we close the issue? I wouldn't recommend creating videos larger than, say, 1 hour.

jstarne1, 2017-08-10:
Ok , yes I did a couple short test videos and played them back, it appears to work properly now.

closed 2017-08-10 by ccrisan