motioneye-project/motioneyeos #1066: 2 FPS

xCozmox, 2017-08-09:
Hello,
I just installed motioneyeos on a Raspberry 3 B and it worked out of the box. First of all... amazing? Thanks!

But as you see in the attachment it makes only 2 FPS.
I assume this is not normal. I use the following camera: 
Raspberry Pi Camera Board v1.3 

I did not change any settings. Can you tell me how to achieve a better quality of the Stream?
On the settings on the side nothing is changing the quality.

Thanks for your help. 

[image]

[image]

ccrisan, 2017-08-09:
Enable Advanced Settings and then increase the framerate setting in Video Device.

closed 2017-08-09 by ccrisan

xCozmox, 2017-08-09:
OMG....thanks for your help!!! ... Sorry that I didn't found this on my own ...

If I put it to 1920x1080 I get arround 10 - 20 FPS. 1280x1024 gives me a bigger view on the room and the FPS is between 20 - 30 FPS.

xCozmox, 2017-08-10:
Hello ccrisan,

Thanks again for this great tool! It works really awesome. 

A question which I have:
1. Why is the viewport with my 1920x1080 smaller on the camera which I have ( I see a smaller part of the room compared to 1280x1024) ? Do you have any idea? Is the camera: Raspberry Pi Camera Board v1.3 maybe not real 1920x1080. I thought that the viewport should be bigger with 1920x1080 compared to 1280x1024... maybe I understand something wrong.
2. Why do I get only 10-20 FPS with 1920x1080. Is this somehow releated to the raspberry pi 3b ? Does it not have enough power? Is it related to motion?

Thanks a lot!

ccrisan, 2017-08-10:
1. Because of the difference in the aspect ratio. You have to cut parts of the image when the aspect ratio differs, in order not to distort the image. However this is a PI Camera question rather than a motionEyeOS one.
2. If you get above 10 fps @ 1920x1080 with motionEyeOS on a RPi, consider yourself lucky. We don't have any kind of hardware acceleration and given the amount of processing that's being done in the CPU, you shouldn't expect anything higher than that.

xCozmox, 2017-08-10:
Thanks a lot!