motioneye-project/motioneyeos #1067: Access away from home?

hattontown, 2017-08-10:
Hello everyone. I feel like I'm missing something obvious. How can I access MotionEyeOS's web portal away from my home network? Or should I just rely on delivery of photos to Google Drive? Thank you.

jasaw, 2017-08-10:
You'll have to enable port forwarding on your router to your motionEyeOS device.

hattontown, 2017-08-10:
Thank you, jasaw. I don't know how to do this, but there seems to be extensive documentation on portforward.com. Seems like the first step is going to be to give my Motioneyeos device a static IP, and then configure that static IP for forwarding.

Thanks a bunch!

closed 2017-08-10 by hattontown

Chiny91, 2017-08-10:
Yes, port forwarding through a router to the motionEyeOS device will work but it is debatable whether this is secure.   Attracting web traffic, especially if the http port is left unchanged, to a humble SBC sounds risky.   Noted that motionEyeOS itself  tends to be updated frequently which will help avoid problems.   If the SBC creaked, a home LAN could be compromised.

Even if the SBC survives, some cameras may have unwanted issues ([example](https://ipvm.com/reports/hik-backdoor)) making matters worse.

Safer to use OpenVPN or perhaps another router as a DMZ if access is required, but stick with Google Drive if not.   Ideally, any IP cams should be firewalled at the router from the net.