motioneye-project/motioneyeos #1070: Recommended Overclocking

hansen-m, 2017-08-10:
I'm running release 20170710 on a Raspberry Pi 1 Model B and it has been perfectly stable. The other day I decided to bump the overclocking setting to 'high' and initially after a reboot it appeared that everything was running fine but it has since become unresponsive. I've read a few other issues that seemed to be related to overclocking instability.

So my question is - what is the recommended stable overclocking setting? And should there be page added to the wiki to help document this?

Thanks!

jasaw, 2017-08-11:
How much you can overclock the Pi is highly dependent on many factors.
- Power supply
- Silicon chip binning
- Environmental factors

How much you can overclock on one device may be very different from the next. I'm not sure if anyone can confidently recommend an overclock setting and guarantee it will be stable.

ccrisan, 2017-08-11:
@hansen-m indeed, the amount of possible overclocking varies from unit to unit. The only "guaranteed" overclocking setting is the default one. In any case, this is not a motionEyeOS issue.

closed 2017-08-11 by ccrisan

hansen-m, 2017-08-11:
@jasaw @ccrisan Thank you for that information. I'll try out a few other settings.