motioneye-project/motioneyeos #1071: Bootloop on raspberry pi zero  w

marcething, 2017-08-10:
Hi ,

i wanted to setup motioneyeos on a Raspberry Pi Zero W, and manually setting up Pi WiFi using wpa_supplicant.conf.
this is my wpa_supplicant.conf file:

country=BE
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
 scan_ssid=1
 ssid="devolo-f4068d3079a9"
 psk="************************" ( I entered the correct one in my file :-) )
}

When I flash the motioneyeos-raspberrypi-20170710.img file to a SD card and add the wpa_supplicant.conf file, I put the card into the Pi zero and start it with the monitor connected , to see what happens.
It starts and after setting root password : done, it says "configuring WIRED network " and then after a few seconds: no device.
How can I  say to not look for a wired ( since there is none ) but a wifi network card?

marcething, 2017-08-10:
Thks for the tip, it was in the wrong directory ;
Works fine now

jasaw, 2017-08-11:
Added troubleshoot steps to [wiki page](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting-Raspberry-PI-Models-A-(Plus)-And-Zero).

closed 2017-08-11 by jasaw