motioneye-project/motioneyeos #1072: Cameras keep going offine

Wycombegav, 2017-08-11:
Hi

Does anyone know why my camera seem to time out and go offline within MotionEyeOS? They are fine after a reboot then after a minute or so they all change the the camera with the line through it. Is there a reason for this? If i reboot they are back to working fine.

ccrisan, 2017-08-13:
Please **attach** your logs.

Wycombegav, 2017-08-14:
Please see attached

[Logs.zip](https://github.com/ccrisan/motioneyeos/files/1222285/Logs.zip)

ccrisan, 2017-08-14:
I see many error lines in your motion log:

    [2:ml2] [INF] [NET] netcam_init_jpeg: jpeg_error 0
    [2:ml2] [INF] [NET] netcam_image_conv: jpeg_error 0

I'd recommend:
1. Try to store media files locally (not on the network share).
2. Report a [motion issue](https://github.com/ccrisan/motioneyeos/wiki/Report-Motion-Issue).

b0rnwr0ng, 2017-08-18:
I believe I have similar issues afyer upgradin to 29170707..  Wifi appears to be dropping on pi3

elyobelyob, 2017-08-23:
I have a new implementation on a pi3 and it's up and down like a yo-yo ... never had any issues before on same hardware (this is a complete rebuild over an existing system).
[Archive.zip](https://github.com/ccrisan/motioneyeos/files/1245318/Archive.zip)

b0rnwr0ng, 2017-08-23:
Same here, one of three was reimaged from scratch as the upgrade corrupted the sd card. I tried lowering image quality to no avail.

jasaw, 2017-08-24:
Does it have something to do with Wifi power management? I was getting wifi patchy connection issue with Raspbian at some point, and turning off wifi power management fixed it.

Maybe try this command to turn off power management.
`iwconfig wlan0 power off`
_I haven't tried this command on my Pi3 because my Pi3 is not operational at the moment._

ccrisan, 2017-08-24:
motionEyeOS already incorporates all the possible ways in which WiFi power saving can be disabled.

elyobelyob, 2017-08-24:
To me, it seems as though when the camera is asked to do something power hungry (e.g. large motion detected or trying to view stored images), it loses connection. I also notice that the camera, on refreshing and reconnecting, is back to a boot stage and you can see it light balancing.

elyobelyob, 2017-09-01:
Has anyone gone back to an earlier version successfully? My system is as good as useless at present.

ccrisan, 2017-09-01:
@elyobelyob can you please try the latest 20170827 prerelease, if you don't have it already? You can, in theory downgrade by logging in remotely (via SSH/putty) and running;

    fwupdate upgrade <version>  # fingers crossed :)

elyobelyob, 2017-09-01:
Installed and get same issue. Just have it plugged in at my desk now, plugged into a monitor and a message has come up.
"brownout: low power supply voltage detected".
This is on a different charger than it's normal positioning, however I notice that when it fails in its normal position a big red light goes on on the board. The power supply was no issue previously (a small Apple 5w adapter - so should be pretty reliable (swapped over with another, same. Then swapped with a larger capacity Apple 12w adapter. Same, although perhaps a bit more reliable? Still, won't stay up for more than a minute.).
I don't have a spare pi3 about to swap out the board, but am considering it.

elyobelyob, 2017-09-01:
Oh wait, I didn't use the prerelease, just the latest. I'll try that now.

ccrisan, 2017-09-01:
@elyobelyob so just to be clear, you're running 20170827 and you're seeing the brownout error messages? If yes, do you by any chance have a multimeter/voltmeter at your disposal so we can make sure the voltage at the PI's pins is within limits?

elyobelyob, 2017-09-01:
Apologies, if I'm being detailed, just trying to figure out the issue.
If the pi is at my desk, within a metre of the router, it's more reliable. At this point it's connected to a tv. When in the next room (through one wall), it starts failing. I've just moved the tv through to see if anymore messages appear. Guess what? It's not crashing. Is HDMI boosting the power or something?
All a bit weird, but this is what I'm getting anyway.

elyobelyob, 2017-09-01:
@ccrisan Yes, that's the version I'm using, and am getting brownout messages. I have a multimeter available, but no idea how to test the pi.
I've unplugged the tv and for some odd reason, the pi now seems stable. It's been rebooted as well. I'll keep watching it.

ccrisan, 2017-09-01:
You need to identify pins 2 an 6 on your PI's header and simply measure the voltage across those pins. You can do a full reboot cycle and see how the voltage varies.

b0rnwr0ng, 2017-09-05:
Im using prerelease (checked off in settings)  as far as I can tell, same
issues.

motionEye Version 0.37
Motion Version 4.0.1+git37b3595
OS Version motionEyeOS 20170827

elyobelyob, 2017-09-06:
I can't really get to my Pi easily to monitor, but definitely seeing restarts.

[root@meye-front ~]# uptime
 19:10:20 up 13 min,  load average: 0.78, 0.73, 0.47

norey, 2017-09-22:
I am having the same issue just today, all was working fine previously. Camera just stops working. It works again after a reboot but then stops after some minutes. I checked dmesg when it stops

`
[  +4.959935] bcm2835-v4l2: error 0 waiting for frame completion
[  +4.960011] bcm2835-v4l2: error 0 waiting for frame completion
[Sep22 15:13] bcm2835-v4l2: error 0 waiting for frame completion
[  +5.040024] bcm2835-v4l2: error 0 waiting for frame completion
[  +4.959988] bcm2835-v4l2: error 0 waiting for frame completion
[  +4.960014] bcm2835-v4l2: error 0 waiting for frame completion
[  +4.959971] bcm2835-v4l2: error 0 waiting for frame completion
[  +4.000005] bcm2835-v4l2: error 0 waiting for frame completion
[  +3.999983] bcm2835-v4l2: error 0 waiting for frame completion
[  +4.000012] bcm2835-v4l2: error 0 waiting for frame completion
[  +5.040005] bcm2835-v4l2: error 0 waiting for frame completion
[  +4.959994] bcm2835-v4l2: error 0 waiting for frame completion
`

What could be the issue here?