motioneye-project/motioneyeos #1077: Motion Die Not Wirkung correctly in version   20170710

ssandel, 2017-08-13:
Hi i got 2 Pi Zero w boards both worked well  with previous versions.
Since

motionEye Version | 0.36.1
-- | --
Motion Version | 4.0.1+git37b3595
OS Version | motionEyeOS 20170710

Motion ist not Well captured ....with Auto noise detection nothing happend.
Sincerely
Sandra

ccrisan, 2017-08-13:
Your issue doesn't make sense. Please use only English words and clearly state your problem.

jasaw, 2017-08-22:
Please describe the problem you're having. Provide log files if you can.
Without any information provided, I suggest you re-flash your microSD cards with the latest MotionEyeOS image.

jasaw, 2019-07-12:
Closing issue due to lack of follow up from issue author.

closed 2019-07-12 by jasaw