motioneye-project/motioneyeos #1079: Running a bash script in the background

Un5een, 2017-08-14:
So I wrote a bash script on _data/etc/monitor_1_ which checks the status of a switch via GPIO and sends a notification if the switch is open.  
Code works perfectly fine but since it's in "_monitor_" script only runs when I`m accessing the GUI.
I tried using _userinit.sh_ but no luck as the code has to keep running to check the GPIO status.  

My noob question is, how do I write get a bash script to run continuously in the background?

Chiny91, 2017-08-14:
What are you doing here; ssh'ing in on a terminal and running bash ?   Not sure where the GUI comes in, nor exactly what you mean by background if you are using _userinit.sh_.

Ending a bash command with `&` backgrounds the command but the command will end when you close the remote session. 

Starting the command with `nohup` allows you to close a remote terminal, without terminating the command.

The motionEyeOS [wiki](https://github.com/ccrisan/motioneyeos/wiki/Tweaks#adding-custom-startup-commands) explains how to run a command at start up but I suspect you have already done that.

Does the log file provide clues ?

Un5een, 2017-08-14:
First of all, sorry for being vague.  
What currently works:
"/data/etc/monitor_1"
a GPIO pin set as input, check status of the input, if the condition is met (switch is closed), overlays it on video feed and sends a notification to my phone.  
But as far as I can understand, "monitor" is only run when you`re looking at the video stream since it's usually meant to (for example) display system temperature overlaid on the video stream.
This means that if i'm not looking at the stream (GUI) I don't get notifications when switch is closed.  

I did try to follow the wiki but had no success.

How can I have this sort of script running continuously? 
(I can paste the script if it makes it easier)

jasaw, 2017-08-15:
What does your script do exactly?
Read GPIO input, if GPIO level condition is met, it does something, and does it quit? If condition is not met, it quits anyway?
In that case, you can run it as a [cron](https://github.com/ccrisan/motioneyeos/wiki/Tweaks#running-commands-at-regular-intervals-cron-support) job.

If your script loops continuously, then you might want to run it in the background. One way of doing it is by using start-stop-daemon. I've whipped up an example of /etc/init.d/S99monitor_1 script that uses start-stop-daemon to run it in the background. Use chmod to make it executable. You start the script by running `/etc/init.d/S99monitor_1 start`. Remount root as read-write before you do any of this stuff because the root filesystem is mounted as readonly.
_**Warning**: None of this stuff has been tested. Just off the top of my head._

/etc/init.d/S99monitor_1
```bash
#!/bin/bash
#
# Starts monitor_1 program.
#

# Make sure the progam exists
[ -f /usr/bin/monitor_1 ] || exit 0

start() {
 	echo -n "Starting monitor_1: "
	start-stop-daemon -S -b -q -p /var/run/monitor_1.pid --exec /usr/sbin/monitor_1
	echo "OK"
}
stop() {
	echo -n "Stopping monitor_1: "
	start-stop-daemon -K -q -p /var/run/monitor_1.pid
	echo "OK"
}

case "$1" in
    start)
        start
        ;;

    stop)
        stop
        ;;
    
    restart)
        stop
        start
        ;;

    *)
        echo "Usage: $0 {start|stop|restart}"
        exit 1
        ;;
esac

exit 0
```

Un5een, 2017-08-15:
I would like the script to loop continuously in the background.  
I`m using motioneyeos for a peep-hole camera but I also want it monitor when my letterbox is opened.  

I`ll look into using start-stop-daemon.  
Thanks for pointing me in the right direction.  
(total noob here so it would be great if there was an easier solution)

Chiny91, 2017-08-16:
An easier solution... well, did you try the [Wiki](https://github.com/ccrisan/motioneyeos/wiki/Tweaks#adding-custom-startup-commands) method ?  That would mean adding a line to your newly created _/data/etc/userinit.sh_ saying `/data/etc/monitor_1 &` 

It might be worth echo -ing any output, or at least errors, if only on a temporary testing basis, whichever method you use, so you can be sure that your script is behaving properly.

Un5een, 2017-08-16:
@Chiny91 that apparently only runs the script once. 
Since the script needs to check the status of a switch continuously, I do not think userinit.sh is appropriate.

Chiny91, 2017-08-16:
I see 3 ways to handle _continuously_ as you are attempting:
(1) Use interrupts; the R Pi can certainly handle interrupts although I suggest a _total noob_ (your words) would be advised to avoid their use.   Efficient and effective,
(2) As per @jasaw, run your script as a cron job, every minute is the most frequent available,
(3) Use your script to do its work as a loop, perhaps using _sleep_ (if available in this Busybox).

You will need to consider that letter box signal, does it latch ?   If not, how long does the signal persist ?   Debouncing the signal could well be necessary.

Perhaps motionEyeOS, as a specialised camera system, is not best suited to monitoring a letter box.

jasaw, 2019-10-30:
Closing issue due to lack of follow up.

closed 2019-10-30 by jasaw