motioneye-project/motioneyeos #1080: Reduce bcm2835-v4l2 max_video_width and max_video_height to reclaim some memory

jasaw, 2017-08-15:
On a Raspberry Pi, there is no point in setting high max_video_width and max_video_height because it's not capable of encoding, or storing, or transferring via network 2592 x 1944 resolution video.

Reduce the default max_video_width and max_video_height to reduce GPU memory usage so more can be allocated to CPU to reduce the OOM events.

Consider making this a config option?

ccrisan, 2017-08-15:
As far as I remember, those settings (also) affect the captured region. In stills mode the captured region is (used to be) smaller. If that's not (no longer) the case, I'm more than happy to remove them.

jasaw, 2017-08-15:
I'm not sure if the captured region has changed recently.

From my observations, by setting max_video_width=1920 max_video_height=1200, we can save slightly more than 20MB of GPU memory at various resolutions. That is a lot of memory for devices with 512MB or less RAM.

Is it worth making this into a config item?

ccrisan, 2017-08-15:
I believe it's safe to limit them to 1920x1080 or similar. That's a fairly high resolution for motionEye on a Raspberry PI :)

jasaw, 2017-08-18:
merged.

closed 2017-08-18 by jasaw