motioneye-project/motioneyeos #1084: FTP media upload

SterlingKG, 2017-08-17:
Test succeeded but no media is actually being uploaded to the server. Any tips and tricks?

ccrisan, 2017-08-17:
Do you have the files saved locally?

SterlingKG, 2017-08-17:
yes, I do. I am able to log in as an FTP client and view them. But my goal is to upload them to my FTP server.

ccrisan, 2017-08-17:
In that case, please enable debugging (in Expert Settings), make sure you generate some motion so that you trigger the creation of the media files and then **attach** your motioneye.log file, downloadable from the Expert Settings as well.

SterlingKG, 2017-08-17:
ok let me try that......

SterlingKG, 2017-08-17:
I am able to view the log but unable to attach it to the image that is supposed to being sent out to my FTP server. Nothing received yet on the server after enabling debugging .

ccrisan, 2017-08-17:
> I am able to view the log but unable to **attach it to the image that is supposed to being sent out to my FTP server**. Nothing received yet on the server after enabling debugging .

You should attach it here, so that I can see it, not to the image. That doesn't make any sense.

SterlingKG, 2017-08-17:
updated the interval to 20 secs. It was set to 900 secs.

SterlingKG, 2017-08-17:
I wonder it is a firewall issue for outgoing rules.

ccrisan, 2017-08-17:
If you won't upload your motioneye.log file here, we'll never know.

SterlingKG, 2017-08-17:
2017-08-17 18:36:03: [motioneye]     INFO: hello! this is motionEye server 0.36.1
2017-08-17 18:36:04: [motioneye]     INFO: main config file /data/etc/motion.conf does not exist, using default values
2017-08-17 18:36:05: [motioneye]     INFO: cleanup started
2017-08-17 18:36:05: [motioneye]     INFO: wsswitch started
2017-08-17 18:36:05: [motioneye]     INFO: tasks started
2017-08-17 18:36:05: [motioneye]     INFO: mjpg client garbage collector started
2017-08-17 18:36:05: [motioneye]     INFO: smb mounts started
2017-08-17 18:36:05: [motioneye]     INFO: server started
2017-08-17 18:36:08: [motioneye]     INFO: adding new camera with id 1...
2017-08-17 18:36:08: [motioneye]     INFO: main config file /data/etc/motion.conf does not exist, using default values
2017-08-17 18:56:20: [motioneye]  WARNING: 403 GET /login/?_=1502996180329&_username=_&_signature=d43c387011c83459784d2b6d40780d43d1e2f405 (192.168.6.38) 15.53ms
2017-08-17 18:58:24: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1502996304358&_username=admin&_signature=aa38e90dfc04cda9002b1e786896486d895708b8 (192.168.6.38) 22.47ms
2017-08-17 18:58:25: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1502996305450&_username=admin&_signature=dbe03cdd8b27f88fd741cad057b9853be31ae7cc (192.168.6.38) 20.56ms
2017-08-17 18:58:26: [motioneye]     INFO: rebooting
2017-08-17 18:58:26: [motioneye]     INFO: executing "/sbin/reboot"
2017-08-17 18:58:27: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1502996306536&_username=admin&_signature=0d493480c5159a657149582d6c768ab3ca0fe1d1 (192.168.6.38) 49.44ms
2017-08-17 18:58:27: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1502996307625&_username=admin&_signature=bac8def062a539df83c82c5ed15a1feb62758c58 (192.168.6.38) 54.39ms
2017-08-17 18:58:29: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1502996308712&_username=admin&_signature=6670507027b79da0759e099e489166df89122d57 (192.168.6.38) 70.87ms
2017-08-17 18:58:30: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1502996309796&_username=admin&_signature=17c34ccc1c48b0f558c948e71370c6d2a7ed508b (192.168.6.38) 41.14ms
2017-08-17 18:58:31: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1502996310880&_username=admin&_signature=ab32e420fed0bd26a4fef5de8a89d9f808e4a971 (192.168.6.38) 41.28ms
2017-08-17 18:58:32: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1502996311963&_username=admin&_signature=8f1dd5faa5937088cffbfe0d9c4d90fb8fd4d9d5 (192.168.6.38) 40.87ms
2017-08-17 18:58:33: [motioneye]     INFO: interrupt signal received, shutting down...
2017-08-17 18:58:33: [motioneye]     INFO: server stopped
2017-08-17 18:58:33: [motioneye]     INFO: tasks stopped
2017-08-17 18:58:35: [motioneye]     INFO: motion stopped
2017-08-17 18:58:35: [motioneye]     INFO: smb mounts stopped
2017-08-17 18:58:35: [motioneye]     INFO: bye!
2017-08-17 18:59:07: [motioneye]     INFO: hello! this is motionEye server 0.36.1
2017-08-17 18:59:11: [motioneye]     INFO: cleanup started
2017-08-17 18:59:11: [motioneye]     INFO: wsswitch started
2017-08-17 18:59:11: [motioneye]     INFO: tasks started
2017-08-17 18:59:12: [motioneye]     INFO: mjpg client garbage collector started
2017-08-17 18:59:12: [motioneye]     INFO: smb mounts started
2017-08-17 18:59:12: [motioneye]     INFO: server started
2017-08-17 18:59:12: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1502996352170&_username=admin&_signature=089d7ba2d998a15c0af15320a7caf732136b49b3 (192.168.6.38) 43.68ms
2017-08-17 18:59:13: [motioneye]  WARNING: 403 GET /config/0/get/?_=1502996353251&_username=admin&_signature=9c01108440b433a6099f35158f59db95e9daebb5 (192.168.6.38) 53.55ms
2017-08-17 18:59:13: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1502996353257&_username=admin&_signature=93b2ed8fe893672940e4c40649c40483c4cae2b3 (192.168.6.38) 40.22ms
2017-08-17 18:59:14: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1502996354340&_username=admin&_signature=de9a752d0ccc9f98d577357638935f3cc52bfd1e (192.168.6.38) 55.09ms
2017-08-17 18:59:15: [motioneye]  WARNING: 403 GET /config/0/get/?_=1502996355334&_username=admin&_signature=31eb2651f8c2696b370fbedee05f0626cc3f9de1 (192.168.6.38) 21.37ms
2017-08-17 18:59:15: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1502996355425&_username=admin&_signature=6118a970f91c639c002fe1b9df4c3fd31c52665d (192.168.6.38) 21.58ms
2017-08-17 18:59:16: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1502996356513&_username=admin&_signature=365559404172de9bdc8c919f292670701b34d024 (192.168.6.38) 21.68ms
2017-08-17 18:59:17: [motioneye]  WARNING: 403 GET /config/0/get/?_=1502996357370&_username=admin&_signature=31f5145e3812bca37a42afbe52680a5c0caad35b (192.168.6.38) 20.91ms
2017-08-17 18:59:17: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1502996357601&_username=admin&_signature=390d8537d3d0d0c8dffeb8babd54ea2e400fec15 (192.168.6.38) 22.87ms
2017-08-17 18:59:18: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1502996358689&_username=admin&_signature=9597af198c28a0a32906ed2e9eaef5715f391890 (192.168.6.38) 22.15ms
2017-08-17 18:59:19: [motioneye]  WARNING: 403 GET /config/0/get/?_=1502996359402&_username=admin&_signature=a42e94c6f5d42c52cf0e8fe6bcf7a6ce3eaf8f5b (192.168.6.38) 23.61ms
2017-08-17 18:59:20: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1502996359776&_username=admin&_signature=6b8bf1a6f97a2560e39563cc5acc5cf3315d56ec (192.168.6.38) 20.95ms
[... cut ...]
2017-08-17 20:30:31: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 18.36ms
2017-08-17 20:30:36: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 14.91ms
2017-08-17 20:30:42: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 17.18ms
2017-08-17 20:30:47: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 16.05ms
2017-08-17 20:30:52: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 17.44ms
2017-08-17 20:30:57: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 17.65ms
2017-08-17 20:31:00: [motioneye]    DEBUG: serving log file "motioneye.log" from "/var/log/motioneye.log"

SterlingKG, 2017-08-17:
I figured it out. It does not support FTPS. I switched it to FTP and went in just fine. Am I correct about FTPS not being supported?

ccrisan, 2017-08-17:
Even though I insisted a couple of times that you **attach** the log file, you pasted it, making it very difficult to read. Anyways, you are correct, there's no FTPS support.

sommo, 2018-11-12:
I have the same issue.. motioneye can create the folder via FTP (no SFTP) and can't test the service successfully.
but no movie will be uploaded, only folder.
i can see that movie files are correctly created into rpi folder but they are not be upload to the ftp server (local server nas).
all settings are correct (seems), ftp setting, triggered motion event and record movie by motion.
I am running motionEyeOS version:

motionEye Version | 0.39.2
Motion Version | 4.1.1
OS Version | motionEyeOS 20180627

on Raspberry PI 3B+
with Network Camera
connection WiFi & Ethernet
[motioneye-3.log](https://github.com/ccrisan/motioneyeos/files/2571564/motioneye-3.log)

i enabled debug and here's in attach the motioneye log

from the log it says that it has correctly uploaded the file but is not, no file at all.
really don't know where is the problem.
Thank you.