motioneye-project/motioneyeos #1086: Can we have a more precise motion detection ?

Nico53220, 2017-08-19:
Hello, in first thank you very much for this wonderfull and helpfull OS ;-)
I just wanna ask something. Can we have a more preicise detection ? 
Because i have a large field of view and when something is moving, it's don't detect by the camera, maybe 0.01 or 0.05%, must be better :-)

Thank you very much .
Nico from France :D

Nico53220, 2017-08-20:
Because i'm using the motion detecion at 0.1% and 1920x1080 . But when a person is moving, he is detected only before 5m, after, the detection don't works. In my opinion the frame changes is too low, in my case 0.01% or 0.05% may be better. 
Is it possible ?

Thank you 😉

jasaw, 2017-08-20:
Have you tried adjusting the "minimum motion frames"? Try lowering it.
What frame rate are you running?

jasaw, 2017-10-11:
@Nico53220 Has the issue been resolved?

davimas, 2017-10-23:
I have a Screensize 1600*1200 with 1,6% Treshold and Minimum Motion Frames 2
Works perfekt to 10 Meter.......

Nico53220, 2017-11-02:
@jasaw  Sorry i forgot to answer but no, the probelm hasn't been resolved . I'm issuing the camera with the minimum motion frame. I think it's one frame . But like @davimas after 10 meter the motion is not detected :-(  Thanks for your answers guys (thanks to you again @jasaw ).

jasaw, 2017-11-02:
@Nico53220 Try turning off "Auto Noise Detection", "Despeckle Filter", and lowering the "Frame Change Threshold". If it still doesn't pick up the motion at 10m, then it's the motion detection algorithm that's the culprit. Might be worth reporting an issue in motion project at that point.

jasaw, 2017-11-14:
This is really a motion software question. You might get more help if you post your question in their mailing list.

closed 2017-11-14 by jasaw