motioneye-project/motioneyeos #1090: video glitches, see the sample (USB cam, RPI3)

remizik, 2017-08-21:
Hello,

I'm using the last version of motionEyeOS for the rpi3 (motioneyeos-raspberrypi3-20170710.img).
everything is working except I get these weird glitches once in a while : 

https://gfycat.com/GiantLawfulGreatdane

I noticed it happens mainly during the day and i'm suspecting an excessive temperature of the board.
Could it be it ?
I also noticed that decreasing the FPS of the camera reduces the amout of "glitching".
My Pi3 is in an enclosure but has aluminum heatsinks and a tiny 5v fan above.

Anyone else experiencing those glitches ?

thanks

jasaw, 2017-08-22:
It might not be a temperature problem. Depending on which video format the USB camera is sending out, a dark scene like night time might compress better, so less data to transfer, less power required to transfer, therefore no problem. I would check the Pi power supply first. I would also test with a powered USB hub in between Pi and camera, see if it makes any difference.

remizik, 2017-08-22:
i'm using a samsung phone charger so 5v/1A.
testing with an active USB hum might be a good idea ! I'll try that ASAP and will update this thread. Thanks

remizik, 2017-08-26:
got a powered usb hub, same behavior ..
a more powerful board like the odroid ux4 would be better I suppose ?

jasaw, 2017-08-28:
Try reporting this issue to [Motion](https://github.com/Motion-Project/motion/issues).
[Here's](https://github.com/Motion-Project/motion/wiki/motioneye) how to report a motion issue.

Motion developers may be able to tell you if it's a motion software bug.

remizik, 2017-08-28:
thanks,
I also tried this : 
a fresh Debian 9 install on a intel core i5 laptop, installed motion, set it up.
video are much smoother now. Can't really reach 30fps on a single cam though, which is REALLY disappointing :( 
But glitches are really rare, and I guess with a slower framerate than 30fps, they would be gone for good.
I think i'll stay on this solution.

I wonder how people use motioneyeos on their RPIs ... unless you use a single cam with a poor resolution/framerate, I really don't see how you can make a proper video surveillance system.

jasaw, 2017-08-29:
@remizik I'm running multiple Raspberry Pi Zero W with Pi Camera, each recording motion at 1280 x 720 resolution 20 fps. They're all running experimental motionEyeOS with hardware accelerated video encoding (currently merged into rpi-omx branch), which will be released when enough testing is done.

This setup seems to work quite well for me, but I'm still tweaking and improving it. I'm designing custom PCB that will give me software controlled IR-cut filter and IR-LEDs, and some other goodies. :-)
On the software side, motion has NEON optimization coming for Pi2 and Pi3, which improves performance. There's so much more software optimizations that can be done.

To me, this is better than a "proper" video surveillance system.

If you have ideas on how to further improve the software, get involved. :-)

jasaw, 2017-10-11:
@remizik I think the video glitches are caused by the insufficient USB bandwidth. I'm not sure which video format your USB cam sends out, but 1080p 30fps is quite a lot of data for a Pi to handle. Pi networking hangs off the USB too, so you're sharing bandwidth.

Anyway, this is not a software issue. Closing.

closed 2017-10-11 by jasaw