motioneye-project/motioneyeos #1091: Pi Zero W Optimal Settings

huvarda, 2017-08-22:
What are the fastest settings for a raspberry pi zero w running the latest release?

jasaw, 2017-08-22:
Depends on what you are trying to achieve, whether network camera, or streaming, or recording, or timelapse, or taking images. There's no one optimal configuration.

My Pi Zero W is recording motion to mp4 file and uploading it to my FTP server that syncs to my Dropbox, so my optimal config is 1280 x 720 resolution, but medium movie quality to keep file size small to optimize for file transfer and Dropbox storage.

Question is, what are you optimizing for?

BTW, this project issues page is for reporting bugs or problems with software.

huvarda, 2017-08-23:
What are the best settings for streaming?

jasaw, 2017-08-24:
It really comes down to the available network bandwidth. It depends on:
- Wifi signal strength between your Zero W and your access point. Poor wifi signal will give you low transfer rate, even though you still can ping it.
- Amount of traffic on the 2.4GHz wifi channel. If you have devices streaming movie, or other MotionEye devices, obviously the available bandwidth is shared across all the devices.

Even with multiple Zero W's in the same house, they may end up with very different configuration.

adrian0, 2017-08-28:
Hi Jasaw,
Just wondering how you are transferring the video from the Pi ZeroW?
I was thinking of writing a script to periodically FTP the video files to another location (and, like you...sync that location to dropbox)

I was thinking of just writing a shellscript to do that.

Regards,
-Adrian

jasaw, 2017-08-29:
@adrian0 On the motionEye web UI, under **_File Storage_**, enable **_Upload Media Files_**, then you can choose **_FTP Server_** as your **_Upload Service_**. Enter your FTP server details and login credentials.

I haven't got time to verify this, but it appears that there's no retry mechanism if the upload fails. Maybe @ccrisan can confirm this?

ccrisan, 2017-08-29:
Indeed, there is no retry mechanism in case the upload fails.

closed 2017-08-29 by ccrisan