motioneye-project/motioneyeos #1093: More than one camera & Googledrive

ajhuk, 2017-08-23:
Hi All,
Having fun with this app on the latest pi3 and a 16gb sd card.  However I'm tracking three IP cameras and trying to upload media files to googledrive.  Works fine for one camera but then I can't seem to get the authorisation key to work for cameras two or three... anyone else had this problem?  I wondered if it was because google sees the device / app as one 'user' and so doesn't permit multiple upload streams??  Workaround would be three googledrive accounts maybe but that seems a tad overboard.  Or could MotionEyeOS be configured to have an media upload at the app level rather than the camera?
Grateful of any advice pretty please....

hxpaul, 2017-08-31:
I have two pis each uploading to the same google drive account
```
motionEye Version | 0.36.1
Motion Version | 4.0.1+git37b3595
OS Version | motionEyeOS 20170710
```
my file storage settings look like this:
[image]
hope it is something simple that you are missing... I notice uploading stopped when my pi got to storage capacity, so I have to only store images + videos locally for a short time

jasaw, 2017-10-11:
@ajhuk I don't think you can use the same authentication key on all the devices. The devices need to get their own key. Have you tried getting each of your devices to obtain their own individual key?