motioneye-project/motioneyeos #1094: Not recording for the full period configured

adrian0, 2017-08-24:
Hello.
I am runniing the latest MotionEyeOS on a Pi Zero W with a 64Gb SD card.
I have it configured for continuous recording and motion detection off and a maximum movie length of 3600 seconds.

I get one file every hour. Each of these files starts from 48, 49 or 50 minutes past the hour.
But the files are only 35 minutes and 50 seconds in length. (sometimes 35m 51s or 35m 52s)

So, I am missing 25 minutes out of every hour.

Anyone else have this happening?

jasaw, 2017-08-28:
Motion software had a bug where recorded movies playback faster than it's supposed to. This bug has been fixed on master, but hasn't made it into motionEyeOS.

One way to tell if you're affected by this bug is to check the timestamp added into the movie. If the timestamp covers the entire hour, but movie duration is less than 1 hour, then you're affected by this bug.

adrian0, 2017-08-28:
It's weird. The timestamp actually doesn't cover the full hour.
For instance, the avi file with filename "13-49-24.avi' has a starting timestamp of 13:49:24 and finishes 14:30:58 with a duration (according to VLC) of 35:51.

Calculating the duration from the timestamps gives 41:34.

The next file in the series is named "14-49-28.avi" and the starting timestamp is of course 14:49:28
So, in this case, there seems to be a 19 minute gap between one file finishing and the next one starting.

jasaw, 2017-08-29:
I've gone through all my recorded videos and I'm seeing it on one of my longer videos. I only record when there's motion, and it works perfectly when the video is short. On one occasion where the video duration was long, it got cut short unexpectedly and the motion was not recorded anymore.

jasaw, 2017-09-04:
@adrian0 I highly suspect https://github.com/Motion-Project/motion/issues/492 is the culprit for the short recording. I've been through motion's source code and there's no flushing of the codec's buffer.

adrian0, 2017-09-06:
I wonder if recording smaller windows of time (say, half hourly) would be a good workaround.
I'll test asap.

jasaw, 2017-09-06:
@adrian0 I've reworked the encoding code in motion software and added draining of codec. Are you able to test this version that's on my fork?
https://github.com/jasaw/motioneyeos/releases/tag/20170827-drain-codec

adrian0, 2017-09-06:
Should be able to install it tonight.
Full, clean install, right?

jasaw, 2017-09-06:
@adrian0 Yes, clean install from scratch if it's not too much trouble.

adrian0, 2017-09-06:
Issue(?) #1
After making a couple of config changes (strangely - Turning on System
Monitoring and Overclocking to 950MHz) and then restarting the system,
somehow my admin password was lost and I could not log in.
Required an entire re-install.

Anyway, it's up and running now.
Camera is facing my car park (dark and mostly black images until the
outside sensor light comes on).
I'll report back tomorrow night on what was captured.

Thanks.

jasaw, 2017-09-06:
@adrian0 Regarding the password problem, it could be the same as issue #1116. Can you please report there the steps you took for the system to lose your password?

clawwd, 2017-09-07:
@adrian0 Regarding the password, I am having the same issue referenced in the post above [https://github.com/ccrisan/motioneyeos/issues/1116](url), does your password happen to be all caps or have capital letters in it?

adrian0, 2017-09-08:
@jasaw, apologies, I ran out of time. I'll check the recorded video in the next 24 hours for you.

@clawwd, thanks. Password has a single upper case char. the rest of the characters are lower case and some nums.

adrian0, 2017-09-08:
Looking good, @jasaw and @ccrisan !
I just downloaded and reviewed two files

Filename 06:58:12.avi.
Start timestamp = 06:58:18 (6 seconds different to filename)
End timestamp = 07:58:17

FIlename 07-58-14.avi.
Start timestamp = 07:58:18 (4 seconds different to filename)
End timestamp =  08:58:17

The delta between filename and the starting timestamp is not something I'd consider a defect.
Probably some timing issue between a process setting up and the start of recording.

Nice work!
-Adrian

jasaw, 2017-09-13:
@adrian0 Thank you for testing this. I'm trying to merge this fix to upstream motion, but I need to know your configuration. 
Which codec did you use? h264_omx? 
Have you been monitoring the recorded videos? Are they still the correct length after extended period of testing?

adrian0, 2017-09-20:
Hey, @jasaw ,
Apologies for the delay, but I just tried to log in and check. I was jotting down the config settings for you, and I ended up changing my password and rebooting....motioneyeos didn't come back up!
Multiple hard reboots (power cycle) later and it's still down!
Not sure if it's hardware related or not.

Seems I need to re-image the sd card.

I'll get to that in the next few days.
Apologies again.

adrian0, 2017-09-22:
OK, 
All back up and working.
I'm running ...
MPEG4 .avi
1024x768
20fps (though I seem to be getting 40/3.0 (as reported on the image when monitoring is turned on)
Max length configured to 3600seconds

It seems as though its recording full hour long video
I still have the slightly odd timestamp/start/finish time issue though, but that's pretty minor.

While I have your attention....Great work by you guys!
Massive Kudos.

adrian0, 2017-09-22:
Oh, another minor issue with the fork created by @jasaw 

I have 2x MotionEyeOS cameras....

Camera 1 is running @jasaw's fork 
(motionEye Version 0.37, Motion Version, 4.0.1+git9f4f0e99dc50b13d339f3bc9273e80a2308e6f5b, OS Version motionEyeOS 20170827)

Camera 2 = motionEye Version 0.36.1, Motion Version 4.0.1+git37b3595, OS Version motionEyeOS 20170710

From Camera 2 (@jasaw's fork) , I can add Camera 1 and view both video feeds from the Camera 1 dashboard.

However, from Camera 2, I am unable to add Camera 1.
I just get, 'authentication failed'.

jasaw, 2017-09-23:
The authentication problem was fixed after my forked release. 
https://github.com/ccrisan/motioneye/commit/7a8b5d78bee059f98bedd69d5cb3416a44d64c81

adrian0, 2017-11-05:
Hi @jasaw hope you are well.
Did the fix to the recorded video length make it into the main branch?
I am just reviewing fotage take using the latest version....
-motionEye Version | 0.37.1
-Motion Version | 4.0.1+git37b3595
-OS Version | motionEyeOS 20171003

But even though I am configured to record 3600s of video per file, I am only getting 35m 51s (2151s)

Regards,
-A

jasaw, 2017-11-05:
@adrian0 You might want to upgrade to motionEyeOS 20171008, which has all my work merged in.

adrian0, 2017-11-08:
Will do.

Interestingly, my installations which use 20171003 report that they are up to date.
Is this perhaps because 20171003 is the 'main branch'?

Rock on, @jasaw !

jasaw, 2017-11-08:
@adrian0 Your devices are only looking for stable release by default. Version 20171008 is still a pre-release.

jasaw, 2018-02-28:
My codec draining fix has been merged into upstream motion and made it into recent motionEyeOS releases, so shouldn't have this recording problem anymore. Closing issue.

closed 2018-02-28 by jasaw

adrian0, 2018-03-16:
NIce work. Thanks @jasaw.
Downloading and installing it now