motioneye-project/motioneyeos #1099: Pi Camera doesnt work in wifi mode

rdmitre, 2017-08-26:
Pi camera doesn´t work when raspi is connected by wifi. The connection to router seems to be ok,

sol4ris33, 2017-08-26:
1. When you open your web browser with rasspberry ip, do you see the camera web interface?
2. Your IP config in motionEyeOs is on Manual or Automatic IP?
3. In your Router configuration/DHCP settings, do you see your raspberry device?

rdmitre, 2017-08-26:
Thanks by your answers.

1. I can see the camera web interface.
2. I set my IP confía to manual.
3. I can see my raspberry connected in my router.
4. I can see the image taken by raspberry camera only when lan cable is
connected. When I disconnect it, is impossible to see te camera image.
5. I access the web interface throw the wifi ip.

sol4ris33, 2017-08-26:
What type of camera are you using? With port DSi? Or usb?

> 1. When you open your web browser with rasspberry ip, do you see the
> camera web interface?
> 2. Your IP config in motionEyeOs is on Manual or Automatic IP?
> 3. In your Router configuration/DHCP [...]
Ad6S05SgTGV5oXCC4NKyhs9hApZHrzemks5sb9qugaJpZM4PDXm2>
-- 
Enviado con Gmail Mobile

rdmitre, 2017-08-27:
I am using an raspberry pi camera v2. When the pi is connected by wifi is
impossible to access the pictures and videos taken folder. Appears as not
accessible. But is possible to access to the camera configuration.

jasaw, 2017-10-11:
@rdmitre You said the Pi camera doesn't work, can you please elaborate more?
Can you see the video stream from a web browser?
What is your resolution and frame rate?
Have you tried moving the Pi nearer to your router so you get stronger wifi signal?