motioneye-project/motioneyeos #1100: Notification email failed: [Errno -3] Temporary failure in name resolution

sol4ris33, 2017-08-26:
Hello,

I'm french so sorry for my bad english....
Since i've installed last motioneyeos version (20170710) last week, i can't make email notification working anymore!
It worked like a charm with the previous version!

I configured it like with the previous version, and it do not work anymore :-(

**Email Addresses**
my.email@gmail.com
**SMTP Server**
smtp@gmail.com
**SMTP Port**
587
**SMTP Account**
my.email@gmail.com
**SMTP Password**
_The passwd google give me at https://myaccount.google.com/apppasswords_
**From Address**
_Empty_
**Use TLS** 
_ON_

What did i made wrong this time?

Thanks a lot for your precious help...!

ccrisan, 2017-08-26:
Please also post a screenshot of your email settings. Also make sure your internet connection is working fine on your PI. Try logging in remotely using SSH/putty and pinging e.g. google.com.

sol4ris33, 2017-08-26:
Oh it's ok i found my problem..... My default gateway address was wrong (i had the default value 192.168.1.1, and it was wrong.....)

Thanks!

closed 2017-08-26 by sol4ris33