motioneye-project/motioneyeos #1102: Installing ngrok

alasso, 2017-08-27:
Hi,

I am running motioneyeOS on a raspberry pi zero w and would like to install ngrok to access from internet.
If it is not ngrok may be another solution. 

Does anybody did this to guide me?

Thanks a lot in advance,
Adrian

ccrisan, 2017-08-27:
https://github.com/ccrisan/motioneyeos/wiki/FAQ#i-cant-seem-to-find-apt-get-or-any-other-package-manager-what-gives

This being said, you either compile your own version of the OS or you go with Raspbian.

closed 2017-08-27 by ccrisan

huvarda, 2017-09-07:
You can try port forwarding to access it from the internet.  If you want your own domain then you can also use dynamic dns to access it from a custom domain.  Other than that I don't think there are any other choices.

bkjaya2020, 2021-02-16:
You can use balena 
https://bkjaya.wordpress.com/2021/02/13/how-to-control-motioneye-cctv-system-on-a-raspberry-pi-3-remotely-using-balenacloud/