motioneye-project/motioneyeos #1103: Choosing advanced settings causes password to no longer work

jennib, 2017-08-27:
I changed the password for admin account and all was good.  Rebooted and all is well.

Then I switched advanced options on and could no longer log in.

ccrisan, 2017-08-27:
What version, what board?

jennib, 2017-08-27:
Raspi 3 latest version

jennib, 2017-08-27:
motioneyeos-raspberrypi3-20170710.img

ccrisan, 2017-08-27:
Please try clearing your browser's cache. If that doesn't work, I'll need some more details (screenshots, errors, logs, something).

jennib, 2017-08-27:
I tried two different browsers.

The problem happened twice.  The first time I change both password and advanced slider.  The second I changed only password, then rebooted and was able to log in both the web page and ssh but then changed the advanced slider and could no longer log in.

Since I am in a rush I won’t be able to try again for a few days.

Cheers

jennib, 2017-08-28:
Just want to say thanks.  I'm using MotionEye on Raspbian stretch and it works great.

jasaw, 2017-10-11:
@jennib Is this still an issue for you?

polkyb, 2017-10-15:
Hi,

I'm getting this issue with motioneyeos-raspberrypi3-20171008

I can log in, change passwords etc. but when I go to advanced mode, change IP to static and reboot I can't get back in with the password that worked before the change.
Tried multiple browsers and cleared caches between attempts.

RPiSpy, 2017-10-24:
mmm this has just happened to me. I updated to the latest version, changed the network timeout setting from 40 to 60 and clicked "Apply". The system rebooted but now I can't login using "admin" and my password. I tried in MS Edge and Chrome.

jasaw, 2017-10-24:
@RPiSpy I think you've hit issue #1116.
Please clear your browser cache and refresh the page.
Try logging in using these passwords: `*****`, or the sha1 of the five stars: `133e3d6b775613651eaedbf7e609d244f2f852af`

dernew, 2017-11-05:
Same problem here after upgrading to 20171008 - system rebooted, tried multiple browsers/ machines, no access via browser or SSH, old password not accepted, nor, ***** or sha1 string nor serial number.  Clearing cache does not fix it.    its a Raspi2, it has been running various versions of Meye for over a year.  System still working for view access [i.e. no password] but I can't change anything or get at any logs.
I've read through #1116 and tried everything there.  Annoying problem.  System rock solid up to the upgrade... there is a lesson for me there !

PieterGit, 2017-12-24:
With upgrade motionEyeOS 20170710 to 20171008 I can confirm password has been changed to `*****` (5 times asterisk) when I changed an advanced setting (rpi3)

ccrisan, 2017-12-27:
@PieterGit this is a duplicate of #1116. The solution is to clear your browser's cache and resetting the password (if possible) after a refresh.

closed 2017-12-27 by ccrisan