motioneye-project/motioneyeos #1104: Dont works Motion Notifications

chrissierra, 2017-08-28:
Hello, please help! I set up email notification, http get to an url, and run a bash command. 

But when there are motion, dont has triggered  anything...

I must to config thread.conf or motion.conf?

or give permission to the user?

jasaw, 2017-09-06:
What's your configuration? Specifically, what's your:
- frame rate
- minimum motion frames
- detection threshold

One way to verify that motion has been recognized is by looking at the video stream from the web UI. If the box around the video stream turns red, it means motion is detected.

Also, which version of motionEyeOS are you running? Would be helpful to attach log files too.

chrissierra, 2017-09-06:
Ok, I get it. Can I attach the image where motion was detected?

jasaw, 2017-09-06:
@chrissierra If you mean attaching photos in the email notification, answer is yes you can. Under "motion notifications", "send an email", there's "Attached Pictures Time Span". Set it to a positive non-zero number. I don't know what number works best, you just have to try it yourself. You also need to enable Still Images for this to work.

chrissierra, 2017-09-06:
I want to attach that photo that was taken on the motion, so in the picture will eventually appear  the "intruder" ... Can I ?

jasaw, 2017-09-07:
@chrissierra Configure your "Still Images" "Capture Mode" to "Motion Triggered". This tells MotionEye to only capture images when motion is detected, e.g. intruder. The photos are stored on your motionEyeOS device. You should be able to see the photos via samba, FTP, and SSH. You can even upload your photos to another server: under "File Storage", there's "Upload Media Files".

To attach the photos in the email, go to "motion notifications", "send an email", there's "Attached Pictures Time Span". Set it to a positive non-zero number.

jasaw, 2017-10-11:
@chrissierra Have you resolved this issue?

closed 2017-10-11 by chrissierra