motioneye-project/motioneyeos #1107: Feature request: Playing recorded video on the web UI

kristian, 2017-08-30:
It would be great, if choosing a recording on the UI, insted of simply displaying a larger thumbnail, you could embed a video control, to view the video right on the UI. HTML5 should be able to embed such a video control cross-browser compliant very easily.

Use case: E.g. on mobile phones (iOS) you are unable (without a third party app) to download and view the video file.

jasaw, 2017-08-30:
There's a pull-request for this feature in motionEye: https://github.com/ccrisan/motioneye/pull/218

kristian, 2017-08-30:
Would be a great addition, thanks for linking this pull-request! 👍

closed 2017-08-30 by kristian

ccrisan, 2017-08-30:
Things are not as easy as they seem but I'll try to make this a priority.

kristian, 2017-08-30:
Thanks Calin! I really do appreciate your effort and really like MotionPie. I dropped you something on your PayPal, enjoy a beer on my card. 👍

ccrisan, 2017-08-30:
@kristian thanks :)

WayneKeenan, 2017-08-31:
I really like this and appreciate the time it takes to develop. I recently setup 6 PiZeroW cams with internet access via a MotionEyeOS hub in under 2 hours so I also recently dropped a few beers worth in your PayPal tip jar.  :)
Having HTML5 video playback support, especially  on iOS, is my biggest wish.