motioneye-project/motioneyeos #1108: Motion notifications not attaching images to emails

hxpaul, 2017-08-30:
I have set up email notifications from my motioneyeos setup (raspberry pi zero with official raspberry pi camera) via a gmail account. All good and emails sending. However images are not being attached to the email. I am capturing images successfully. If I extend the "timespan to look for images" to more than one hour then I get images, I suspect these are not the images connected to the motion detection though. It feels like there is an hour's difference, as if the code that looks for the images is in one timezone and the code that sends the email is in another. Is this possible? They should both be in sync, even if the emails were sent with the wrong timestamp (say utc time instead of bst) the images and the email preparation should in the same local time. I've tried changing my timezone from UTC to London time to see if it makes a difference, but no. Here is a sample email:
 
[image]

Any advice here anyone?

Ace repo that I use a lot, thanks.

ccrisan, 2017-08-30:
See https://github.com/ccrisan/motioneyeos/wiki/FAQ#what-should-i-do-to-have-pictures-attached-to-motion-notification-emails.

 Also please try the latest 20170827 prerelease as it incorporates some email attachment improvements.

hxpaul, 2017-08-31:
Thanks, will clear down the image folder.
Is there a way to upgrade to a prerelease from the ui? Says I'm up to date - do I have to download the image manually?

closed 2017-08-31 by hxpaul

ccrisan, 2017-08-31:
@hxpaul https://github.com/ccrisan/motioneyeos/wiki/FAQ#i-noticed-a-new-version-has-been-released-why-does-my-motioneyeos-say-its-up-to-date

hxpaul, 2017-08-31:
awesome thanks 👍