motioneye-project/motioneyeos #1109: Pi Zero(-W) activity and USB dongle LED's not blinking

Scootcha, 2017-08-30:
This is very minor enhancement request but unlike when running Raspbian the Zero 1.3 and Zero W LED stays fully illuminated with 20170827 and earlier versions instead of blinking with activity/load.  The same is true for the USB wireless dongle (both RTL8818 and RT5370 based) LED indicator when using the 1.3.  Instead of blinking with activity like they do with Raspbian they stay fully illuminated eliminating visual troubleshooting and I think generating more heat than necessary.   Other than that, 20170827 runs great and I love the snapshot feature.  Thanks for the work.

ccrisan, 2017-08-30:
How can we turn off those leds? Is this controlled via a config.txt entry?

Scootcha, 2017-08-30:
Thanks, it looks like it would.
https://www.jeffgeerling.com/blogs/jeff-geerling/controlling-pwr-act-leds-raspberry-pi

# Disable the ACT LED on the Pi Zero.
dtparam=act_led_trigger=none
dtparam=act_led_activelow=on

Had to use these commands since sudo was not found

echo none |  tee /sys/class/leds/led0/trigger
echo 1 |  tee /sys/class/leds/led0/brightness

It seems like 1 just turns the LED off.  I have not figured out how to indicate disk or CPU activity like it used to do with Raspbian.  Although upon further inspection the newer Raspbian builds now rarely blink even after killing off motion.

I also haven't found where to control the wireless dongle blinking which is the real priority for me since it heats up more than the tiny onboard Zero one.  I will poke around a bit more.

ccrisan, 2017-08-31:
The `config.txt` used by motionEyeOS is minimal and the default values are (supposedly) used by the firmware (which is also the same version as with the latest Raspbian). I see two (non-exclusive) options here:
 1. preserve `/boot/config.txt` across updates so that whatever changes the user applies will be kept.
 2. use the  `/data/etc/userinit.sh` script to always run the LED commands

As for the WiFi led, isn't there a kernel module parameter to control it?