motioneye-project/motioneyeos #1111: Remote motionEye Camera: authentication failed

angares, 2017-08-31:
Hello!

I'm trying to add a Remote motionEye Camera to motionEye Server, but "authentication failed". Checked password and ip, and no FNC. Don't know what to do now - can you help me?

I'm using latest version of motioneyeos (Pi2 - 20170827) and on server motionEye, installed on Pi3 with raspian jessie.

Greetings!
Nils

[image]
[image]
[image]
[image]

ccrisan, 2017-08-31:
I'll look into it.

WayneKeenan, 2017-08-31:
Maybe as simple as pressing TAB after entering the password?

angares, 2017-08-31:
@WayneKeenan I checked it, thats not the problem 
@ccrisan Thank you very much

petermeigs, 2017-09-02:
I struggled with this authentication failure and finally went back to the 0.35.2 version.  Then that acted up as well, not letting me login in from my browser. As an act of desperation and a wild guess based on seeing a comment about http cookie name prefix in the 20170611 release, I deleted all the cookies and cached files in my browser. Now the 0.35.2 version is working just fine.

I haven't gone back to try 20170827 again for a real test but deleting cookies and cached files might help.

ccrisan, 2017-09-02:
@angares you were right - there was indeed a bug when sending password to a remote motionEye camera. https://github.com/ccrisan/motioneye/commit/1d4041c0ca81c0911a8a20f47b7fae179c2ce167 fixes this. I will release a 0.37.1 version soon. Thanks for reporting this.

closed 2017-09-02 by ccrisan