motioneye-project/motioneyeos #1112: rpi-omx: unable to get c920 camera working

ngarafol, 2017-09-01:
Hi,

I am using rpi-omx branch, image built from source. Was hoping to get logitech c920 working with better framerates. Board is raspberrypi (first one)
But all I get is error: unable to capture first image from motion. No matter what settings I put. Removing and adding camera does not fix the issue.

```
motion.log:
[-1259621496:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[-1259621496:motion] [ALR] [ALL] conf_cmdparse: Deprecated config option "thread" since after version 3.4.1:
[-1259621496:motion] [ALR] [ALL] conf_cmdparse: The "thread" option has been replaced by the "camera" option.
[-1259621496:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
[-1259621496:motion] [NTC] [ALL] motion_startup: Motion 4.0.1+gitd23e263490a54329329b64883f1860ada5e5920b Started
[-1259621496:motion] [NTC] [ALL] motion_startup: Logging to syslog
[-1259621496:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[-1259621496:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1] [ERR] [ALL] motion_init: Error capturing first image

```

Camera works with normal "master" branch release. 

PS: How can I restart motion with debug 8 or 9 option? Tried killing and starting motion from command line but seems like it gets restarted in the meantime :)

ccrisan, 2017-09-01:
@ngarafol thanks for reporting this issue and thanks for taking the time to test the rpi-omx branch. There's still some work to be done on the branch before we can declare it stable enough to be merged.

> PS: How can I restart motion with debug 8 or 9 option? Tried killing and starting motion from command line but seems like it gets restarted in the meantime :)

You either need to stop motionEye completely (`/etc/init.d/S85motioneye stop`) or remove/disable all cameras from motionEye.

ngarafol, 2017-09-01:
Oh, thanks. Actually deleted camera but forgot to check it motion is still active. Tried with debug 8 and 9 but still no extra information. Could it be that some other component is the issue here?

```
[-1260518520:motion] [NTC] [ALL] [Sep 01 13:52:47] motion_startup: Using default log type (ALL)
[-1260518520:motion] [NTC] [ALL] [Sep 01 13:52:47] motion_startup: Using log type (ALL) log level (ALL)
[-1260518520:motion] [DBG] [VID] [Sep 01 13:52:47] bktr_mutex_init: BKTR is not enabled.
[-1260518520:motion] [NTC] [ENC] [Sep 01 13:52:47] ffmpeg_global_init: ffmpeg libavcodec version 57.89.100 libavformat version 57.71.100
[0:motion] [NTC] [ALL] [Sep 01 13:52:47] main: Camera 0 is from /data/etc/motion.conf
[0:motion] [NTC] [ALL] [Sep 01 13:52:47] main: Camera 0 is device: /dev/video0 input -1
[0:motion] [NTC] [ALL] [Sep 01 13:52:47] main: Stream port 8888
[0:motion] [NTC] [ALL] [Sep 01 13:52:47] main: Waiting for threads to finish, pid: 22665
[0:web_control] [NTC] [STR] [Sep 01 13:52:47] http_bindsock: listening on 127.0.0.1 port 7999
[0:web_control] [NTC] [STR] [Sep 01 13:52:47] httpd_run: Started motion-httpd server on port 7999 (auth Disabled)
[1:ml1] [NTC] [ALL] [Sep 01 13:52:47] motion_init: Camera 0 started: motion detection Enabled
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] vid_start: Opening V4L2 device
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_start: Using videodevice /dev/video0 and input -1
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_get_capability: 
------------------------
cap.driver: "uvcvideo"
cap.card: "HD Pro Webcam C920"
cap.bus_info: "usb-20980000.usb-1.2"
cap.capabilities=0x84200001
------------------------
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_get_capability: - VIDEO_CAPTURE
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_get_capability: - STREAMING
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_select_input: name = "Camera 1", type 0x00000002, status 00000000
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_select_input: - CAMERA
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_select_input: Device doesn't support specifying PAL/NTSC norm
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_set_pix_format: Config palette index 17 (YU12) doesn't work.
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_set_pix_format: Supported palettes:
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_set_pix_format: (0) YUYV (YUYV 4:2:2)
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_set_pix_format: 0 - YUYV 4:2:2 (compressed : 0) (0x56595559)
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_set_pix_format: (1) H264 (H.264)
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_set_pix_format: 1 - H.264 (compressed : 1) (0x34363248)
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_set_pix_format: (2) MJPG (Motion-JPEG)
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_set_pix_format: 2 - Motion-JPEG (compressed : 1) (0x47504a4d)
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_set_pix_format: Selected palette H264
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_do_set_pix_format: Testing palette H264 (800x600)
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_do_set_pix_format: Using palette H264 (800x600) bytesperlines 1600 sizeimage 960000 colorspace 00000008
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_scan_controls: found control 0x00980900, "Brightness", range 0,255 
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_scan_controls: 	"Brightness", default -8193, current 128
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_scan_controls: found control 0x00980901, "Contrast", range 0,255 
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_scan_controls: 	"Contrast", default 57343, current 128
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_scan_controls: found control 0x00980902, "Saturation", range 0,255 
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_scan_controls: 	"Saturation", default 57343, current 128
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_scan_controls: found control 0x00980918, "Power Line Frequency", range 0,2 
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_scan_controls: 	"Power Line Frequency", default 2, current 2
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_scan_controls: found control 0x00980913, "Gain", range 0,255 
[1:ml1] [NTC] [VID] [Sep 01 13:52:47] v4l2_scan_controls: 	"Gain", default 57343, current 0
[1:ml1] [DBG] [VID] [Sep 01 13:52:47] v4l2_set_mmap: mmap information: frames=4
[1:ml1] [DBG] [VID] [Sep 01 13:52:47] v4l2_set_mmap: 0 length=960000 Address (b3a15000)
[1:ml1] [DBG] [VID] [Sep 01 13:52:47] v4l2_set_mmap: 1 length=960000 Address (b3815000)
[1:ml1] [DBG] [VID] [Sep 01 13:52:47] v4l2_set_mmap: 2 length=960000 Address (b372a000)
[1:ml1] [DBG] [VID] [Sep 01 13:52:47] v4l2_set_mmap: 3 length=960000 Address (b363f000)
[1:ml1] [NTC] [ALL] [Sep 01 13:52:47] image_ring_resize: Resizing pre_capture buffer to 1 items
[1:ml1] [DBG] [VID] [Sep 01 13:52:47] v4l2_capture: 1) vid_source->pframe -1
[1:ml1] [DBG] [VID] [Sep 01 13:52:47] v4l2_capture: 2) vid_source->pframe -1
[1:ml1] [DBG] [VID] [Sep 01 13:52:47] v4l2_capture: 3) vid_source->pframe 0 vid_source->buf.index 0
[1:ml1] [DBG] [VID] [Sep 01 13:52:47] v4l2_capture: vid_source->buf.bytesused 25567
[1:ml1] [DBG] [VID] [Sep 01 13:52:47] v4l2_capture: the_buffer index 0 Address (b3a15000)
[1:ml1] [DBG] [VID] [Sep 01 13:52:49] v4l2_capture: 1) vid_source->pframe 0
[1:ml1] [DBG] [VID] [Sep 01 13:52:49] v4l2_capture: 2) vid_source->pframe 0
[1:ml1] [DBG] [VID] [Sep 01 13:52:49] v4l2_capture: 3) vid_source->pframe 1 vid_source->buf.index 1
[1:ml1] [DBG] [VID] [Sep 01 13:52:49] v4l2_capture: vid_source->buf.bytesused 1190
[... cut ...]
[1:ml1] [DBG] [VID] [Sep 01 13:52:57] v4l2_capture: 3) vid_source->pframe 0 vid_source->buf.index 0
[1:ml1] [DBG] [VID] [Sep 01 13:52:57] v4l2_capture: vid_source->buf.bytesused 10956
[1:ml1] [DBG] [VID] [Sep 01 13:52:57] v4l2_capture: the_buffer index 0 Address (b3a15000)
[1:ml1] [DBG] [ALL] [Sep 01 13:52:57] mlp_capture: vid_return_code 1
```

I changed motion.conf a little in order to test easier:

```
stream_localhost off
stream_port 8888
height 600
width 800

output_debug_pictures off
setup_mode off
webcontrol_localhost on
webcontrol_html_output on
webcontrol_port 7999
```

jasaw, 2017-09-04:
I suspect this is a motion software issue.
@ngarafol Can you please report this issue to [Motion](https://github.com/Motion-Project/motion/issues)? Motion developers might be able to give us some hints.
[Here's](https://github.com/Motion-Project/motion/wiki/motioneye) how to report a motion issue.

jasaw, 2017-09-06:
@ngarafol Just FYI, hardware acceleration in rpi-omx branch only applies to video encoding. No work has been done to hardware accelerate the video decoding.

jasaw, 2017-10-11:
This looks like a motion bug. I'm not going to fix this one. We'll just have to wait for motion developers to fix this. Alright to close this issue?

jasaw, 2017-11-14:
Upstream motion has marked this issue as resolved. Closing issue.

closed 2017-11-14 by jasaw