motioneye-project/motioneyeos #1113: Unable to log in after disabling Samba and FTP services and rebooting

Imburr, 2017-09-01:
I disabled Samba and FTP services, rebooted. I can pull up Web UI, but neither my creds or default creds work. Same with SSH, both root and admin, with my password and blank password fail.

How can I fix?

ccrisan, 2017-09-02:
1. What version do you run?
2. Did you change your admin password at the same time when you disabled Samba/FTP services?
3. Do you have a Linux machine at your disposal to pull out the SD card and plug it in into your PC? This way you can inspect the logs (to see what happened) and reset your password.

P.S. 20170827 radically changes the way admin password is managed. But this has yet to be tested.

louis141, 2017-09-02:
Hi and good evening.
I have a similar Problem: 
- Install 20170827 from the scratch. (SD Formatter, Win32DiskImager)
- Changes /dev/mmcblk0 pathes for emmc card (see -odroid-xu4 doesn't boot #908)  :)
- run the new image -> ok
- restore the backup file -> ok
- change same parameters -> after some commands: an error has occurred, please refresh
- after refresh no login possible -> invalid credentials 

found in motioneye.log:
 Traceback (most recent call last):
  File "/opt/buildroot/motioneyeos/output/odroidxu4/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1425, in _stack_context_handle_exception
  File "/opt/buildroot/motioneyeos/output/odroidxu4/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1627, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 291, in post
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 200, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 526, in set_config
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 373, in set_camera_config
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 777, in motion_camera_ui_to_dict
KeyError: 'manual_snapshots'
2017-09-01 23:09:46: [motioneye]    ERROR: 500 POST /config/0/set/?_=1504300187771&_username=admin&_signature=248ec66a3a6c7dc4984b878ec77eb4daffbc3508 

I hope it helps... :)

ccrisan, 2017-09-02:
@louis141 I believe this is a slightly different problem.
1 . Did you restore a backup from a different version?
2. Have you tried clearing your browser's cache, refreshing and logging in?

louis141, 2017-09-02:
backup was from 20170710
additional I did try run the 20170827 version without restore. Same problems after some commands and apply:
" an error has .... pls refresh..."
I tried the login from different computers: same result after the error message (no login possible)

jezmck, 2017-09-03:
I've found that after updating to the latest version, and even using the latest image from scratch, a reboot changes/corrupts the admin password such that I can't login again.

ccrisan, 2017-09-04:
@jezmck can you please describe exactly, step by step, what you are doing to get this error? This is important to us, especially if it's happening with 20170827.

jezmck, 2017-09-06:
Honestly, I don't think I can anymore, nor can I rule out SD card corruption and writing difficulties, sorry!

I've been trying to do it using a raspi to write the cards because my windows PC is dying, all while my toddler distracts me.

ccrisan, 2017-09-06:
This might be a duplicate of #1116.