motioneye-project/motioneyeos #1114: Error saving configuration

FabrizioC, 2017-09-03:
Unicodes in SSID cause error saving wpa_supplicant.conf when apply any configuration settings.

My wifi ssid contains a unicode character: "", I'm able to configure the SD and the Raspberry Pi Zero W can connect to the network and receive the IP, I can reboot or shutdown from the motioneye interface, see the camera streaming, etc... until I try to alter a configuration parameter, when I push the apply button an error occurred and after a reboot the Raspberry is unable to connect, and continue rebooting.
Backing up the configuration (after the error, and before reboot) the wpa_supplicant.conf have no network setting:
`
country=IT
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
`

[motioneye.log.txt](https://github.com/ccrisan/motioneyeos/files/1272927/motioneye.log.txt)

ccrisan, 2017-09-04:
I will look into it.

michaelrcarroll, 2018-08-05:
Bump as I'm seeing this too. Initially it connects, but upon changing any settings, the connection configuration is wiped.

tangtony, 2019-06-03:
Any updates on this or even a workaround?