motioneye-project/motioneyeos #1115: Only first few frames of motion recorded

topshed, 2017-09-04:
Not sure if this is the same as #1094 but only the first couple of seconds of a recorded movie contains frames with motion. The rest of the movie is just one frame repeated.  Tried with a Pi B and Pi3 image. Reproducible.

jasaw, 2017-09-04:
Which version of MotionEyeOS are you running?

topshed, 2017-09-05:
20170710

jasaw, 2017-09-06:
MotionEyeOS-20170710 is running very old version of motion software which has known issues with video recording.

A lot of video recording issues have been fixed. Are you able to test this version that's on my fork?
https://github.com/jasaw/motioneyeos/releases/tag/20170827-drain-codec

timdonovanuk, 2017-09-22:
I'll give this a go, as the current MotioneyeOs is unusable on a Pi Zero W, with regards to movie capture. Will report back.

timdonovanuk, 2017-09-22:
Well, I get either exactly the same (garbage move files with 5 seconds of black screen and 1 frame) or "no movies recorded". Sorry.

timdonovanuk, 2017-09-22:
Same problem, half the video is a black screen, the rest is a single still frame. I think the file is also corrupt in some way, VLC certainly has a hard time playing it (you cannot skip ahead or rewind etc). I've attached a sample zipped (github wouldn't let me upload mp4).
[Camera1_15-36-59.zip](https://github.com/ccrisan/motioneyeos/files/1325035/Camera1_15-36-59.zip)

jasaw, 2017-09-22:
@timdonovanuk I can playback your video file without any problem. Can you please try a different player?

It looks like it's 320 x 200 resolution at 1 fps. You might want to try the OMX codec if you're using my forked version to get higher frame rate.

timdonovanuk, 2017-09-22:
I've tried on my usual VLC player but just tried SMPlayer too. It worked in SMPlayer this time...I've had it simply crash from trying to play MotionEyeOs output though. 

Super weird, because VLC is quite widespread and plays pretty much anything. I wonder if its just my machine! 

I'll clear my browser cache and try your fork. Many thanks @jasaw !

timdonovanuk, 2017-09-22:
Thanks @jasaw this is working well for me! I changed the codec type to Matroska OMX mkv and have no problems playing videos now. 

Really weird, as I've never had any problems playing other movie files..

jasaw, 2017-09-25:
@timdonovanuk You may want to try [MX Player](https://mxplayerdownload.co/) if the other players are giving you problems in the future. MX Player has been working well for me.

As for the OMX codec on my fork, it will be available on official motionEyeOS eventually.

closed 2017-09-25 by jasaw

timdonovanuk, 2017-09-25:
To be honest, VLC is the golden standard of media players, and if that struggles to play it, then something is very wrong (albeit, maybe with my computer - although again, never had a single problem with other media).  

I tried watching one of the MP4's generated byMotionEye on my Android phone, and it crashed the stock video player...

topshed, 2017-09-25:
Changing media players doesn't help for me. I've tried QuickTime and even imported into iMovie. Still multiple repeated frames, mostly at the start of the movies.

jasaw, 2017-09-25:
@topshed Which version of motionEyeOS are you running? Are you running my forked version? Honestly, I have never used QuickTime or iMovie, so no idea how well they handle H264.

@timdonovanuk H264 standard has many profiles. Many older versions of players cannot even handle H264 baseline profile properly, especially older smartphones. Try the latest MX Player on your Android phone. I never come across a stock player that is good enough, even on my brand new Android phone. They always struggle with some format, or freeze, or even crash. Which version of VLC are you running?

topshed, 2017-09-26:
I'm convinced this is a big in motion. I've switched to using a Pi 3 with the latest Stretch Raspbian image and the latest version of motion. I still see the same problem (although not as severe). At some parts of the movie, normally the start,  there are multiple repeated frames.

jasaw, 2017-09-27:
@topshed Which versions of motion did you run? Please provide the git hash of the versions that you were referring to, otherwise no one can help you. When you say latest version, do you mean git-2eedfdc01522102f3d5d5b99dcc7fb35db1e8147 ?

I have fixed multiple motion bugs related to movie recording, some merged into upstream, some still pending.

metalshreds, 2017-10-20:
I'm wondering if this is the same problem I'm having. first few seconds record, then freezes for most of the video. rpi3 running latest motion version, version: 4.0.1+git37b3595. 
[Front of House_10-58-10.avi.zip](https://github.com/ccrisan/motioneyeos/files/1402829/Front.of.House_10-58-10.avi.zip)

jasaw, 2017-10-22:
@metalshreds That motion version is very old, from Wed Nov 2 2016. Since then, there has been A LOT of work done to fix video recording and playback issues. Please try motionEyeOS pre-release version 20171008 and report back.

metalshreds, 2017-10-23:
ahh sorry, was not aware I was using such an old version. I'll do that and get back, thank you :)

metalshreds, 2017-10-24:
@jasaw this latest version has been working really well, thank you! videos are very smooth using H.264/OMX :)

jasaw, 2017-10-24:
@metalshreds Very glad to hear that ! :-)