motioneye-project/motioneyeos #1116: password issues with motionEyeOS 20170827

clawwd, 2017-09-05:
I am trying motionEyeOS 20170827 on RPI 1, 2 and running into the units losing their password configured in the GUI once rebooted. I keep getting "Invalid credentials". It doesn't happen at the same number of reboots but once it happens, it is gone. No version of the login including the initial "admin" with no password works. When it happens, SSH and FTP logins have the same issue with logins failing.

Strange thing is I have a main camera configured to use "Network Camera" with one of the units having this issue and the stream has not stopped..yet.

Is this a known issue and there is any way to overcome it?

ccrisan, 2017-09-05:
This might be a known issue but we haven't clearly identified it. There's at least one other user reporting a similar problem, but I have yet to hear some details from him.

In the meantime, can you please try and help us reproduce the problem by lying down some steps?

clawwd, 2017-09-06:
Hardware: RPI2 with USB WiFi, and Microsoft Lifecam 3000 (all connected before configuration begins):
1. Used SDCARD formatter from sdcard.org to format the SD card.
2. Used Win32 Disk Imager to burn a freshly downloaded copy of version 20170827
3. Inserted SDCARD into RPI and connected power/CAT5 Ethernet cable.
4. Opened web browser/GUI up and pointed to the IP pulled by the RIP2 via DHCP (hardwired).
5. Changed timezone, Admin / Surveillance password, and setup WiFi credentials. Saving rebooted the unit.
6. Removed CAT5 cable and powered up the RPI.
7. Reconnected via web browser/GUI (using new admin credentials)to the WiFi IP address. (IP was statically mapped to the units MAC address in my router so the RPI would pull the same IP each time)
8. Added camera in the GUI and and tested functionality. Used default settings.
9. Powered down RPI and mounted/reassembled into camera enclosure.
10. Mounted unit to the side of my house and reconnected to power source.
11. Reconnected via web browser/GUI to WiFi IP address. Could not login using new or old admin credentials. Keeps giving "Invalid Credentials".

* I unmounted the unit and brought it back indoors and still observed the same credential problem.

* I verified SSH and FTP had the same credential problem.

* On one of the units I had the issue with, I performed the above but before powering it down to remount in its permanent location, I got on my hub camera and added this camera as a Network Camera. Once the unit was powered down and remounted, I couldn't log in because of the credential error but upon checking my main camera, the stream from this camera  continued to be active.

* I may open another issue on this but I thought I would mention that even though I put all of my cameras on the same software version, I could only add them to my hub camera as "Network Camera". Even the units that allowed me admin access directly would fail when mapped as "Remote MotionEYE Camera" as it was stating invalid credentials.

ccrisan, 2017-09-06:
@clawwd thanks for the detailed instructions. I'll try to reproduce this and get back to you.

Regarding the Remote motionEye Camera authentication problem, that's a different issue fixed with https://github.com/ccrisan/motioneye/commit/1d4041c0ca81c0911a8a20f47b7fae179c2ce167.

ccrisan, 2017-09-06:
@clawwd do you have any special characters in your password?

ccrisan, 2017-09-06:
@clawwd I have been trying to reproduce your issue for an hour here and I just cannot seem to make it *not* accept the password. Could you try changing your password into something simple and see if it still happens? Also is this fully reproducible on your side?

clawwd, 2017-09-07:
First, thank you for the quick response and the time you spent trying to duplicate.

The password is all letters. The only thing a lil different is there are capital letters in it

Unfortunately it will be the weekend before I can test. The only way I can check an easier password will be to wipe it btw which I will do.

jezmck, 2017-09-07:
I believe this is similar to what I encountered. I originally tried no password at all, then a very simply password of lowercase letters and numbers.

ccrisan, 2017-09-07:
Guys, this really bothers me as I still can't reproduce the problem. Can anyone send me (or attach) their log files after this problem happens? Recovering the logs could be done using a separate Linux machine, given that SSH logins no longer work.

Another thing: when this happens, does the motionEye UI login still work? Or is it also dead?

jbradach, 2017-09-09:
I encountered this issue today, reinstalled, and encountered it again. Login isn't working via SSH or the motionEye UI.

Log attached.

[motioneye-pi-zero-w-20190909.txt](https://github.com/ccrisan/motioneyeos/files/1290129/motioneye-pi-zero-w-20190909.txt)

magaa, 2017-09-10:
Same Here. I have ah Raspberry Pi Zero W with motionEyeOS. When i change the Time Zone the Login isn't working.
- fresh install of motionEyeOs
- first i change my Password
- reboot or not it dosn't matter
- change Timezone and reboot
- no Login posible 
- UI and Camera is working

ccrisan, 2017-09-12:
Guys is everyone facing this problem using Raspberry PI Zero? Does this happen on other models/devices?

ccrisan, 2017-09-12:
@magaa does this happen if you use a simple password (e.g. abcd1234)?

ccrisan, 2017-09-12:
Guys, I have a very hard time here trying to reproduce this problem. Is this fully reproducible on your side? If yes, would anyone of you be willing to help me do some intensive debugging via Hangouts/Skype/whatever?

magaa, 2017-09-12:
Yes, i tried it with a simple Password. My Workaround ist to type in a Password at the same time i change the Timezone.

ccrisan, 2017-09-12:
@magaa is your issue **fully reproducible**? Or does it happen from time to time?

jezmck, 2017-09-12:
I too was changing the timezone.

jonwilliams84, 2017-09-12:
I think I am facing a similar issue with the surveillance user password. I have Basic auth turned on but whenever I try to login to the streaming URL (:8081) I get the Authentication Required box so I type user/password and it justs throws up the password prompt again and again. I have tried an external app and a few different browsers and get the same issue.

therocco, 2017-09-12:
I can reproduce the issue every time I try to change the camera name.

dshults, 2017-09-13:
Running Version 20170827 from Software Update (upgrade) Check button. Symptoms began after this update.
I see two symptoms:
1) Surveillance Username login at the browser GUI fails. Admin login still works fine here.
2) Mapped drives to motionEyeOS shared folders using smb:// from either OpenSUSE or Ubuntu fail.

These have always worked flawlessly in the past.
SSH from a terminal or other utility (Remmina, PuTTY, etc) still work fine with original login/password.

dshults, 2017-09-13:
Here is a small snippet from the suspect log file log.smbd when failing:

[2017/09/12 17:56:13.093214,  0] ../lib/util/charset/codepoints.c:16630(get_conv_handle)
  dos charset 'CP850' unavailable - using ASCII
[2017/09/12 17:56:13.101019,  0] ../lib/util/charset/codepoints.c:16630(get_conv_handle)
  dos charset 'CP850' unavailable - using ASCII
[2017/09/12 17:56:16.725292,  0] ../lib/util/charset/codepoints.c:16630(get_conv_handle)
  dos charset 'CP850' unavailable - using ASCII
[2017/09/12 18:05:50.611571,  0] ../lib/util/become_daemon.c:124(daemon_ready)
  STATUS=daemon 'smbd' finished starting up and ready to serve connections
[2017/09/12 18:05:50.653328,  0] ../source3/lib/util_sock.c:334(open_socket_in)
  **open_socket_in(): socket() call failed: Address family not supported by protocol**
[2017/09/12 18:05:50.653491,  0] ../source3/smbd/server.c:1080(smbd_open_one_socket)
  smbd_open_one_socket: open_socket_in: Address family not supported by protocol
[2017/09/12 18:05:50.693447,  0] ../source3/lib/util_sock.c:334(open_socket_in)
  open_socket_in(): socket() call failed: Address family not supported by protocol
[2017/09/12 18:05:50.693601,  0] ../source3/smbd/server.c:1080(smbd_open_one_socket)
  smbd_open_one_socket: open_socket_in: Address family not supported by protocol
[2017/09/12 18:06:32.571558,  0] ../lib/util/charset/codepoints.c:16630(get_conv_handle)
  dos charset 'CP850' unavailable - using ASCII
[2017/09/12 18:06:32.713436,  0] ../lib/util/charset/codepoints.c:16630(get_conv_handle)
  dos charset 'CP850' unavailable - using ASCII
[2017/09/12 18:06:32.724463,  0] ../lib/util/charset/codepoints.c:16630(get_conv_handle)
  dos charset 'CP850' unavailable - using ASCII
[2017/09/12 18:06:44.842301,  0] ../lib/util/charset/codepoints.c:16630(get_conv_handle)
  dos charset 'CP850' unavailable - using ASCII
[2017/09/12 18:06:44.858600,  0] ../lib/util/charset/codepoints.c:16630(get_conv_handle)
  dos charset 'CP850' unavailable - using ASCII
[2017/09/12 18:06:46.586505,  0] ../lib/util/charset/codepoints.c:16630(get_conv_handle)
  dos charset 'CP850' unavailable - using ASCII

ccrisan, 2017-09-13:
@dshults this is a different issue. There are many parts of motionEyeOS that involve authentication. Please let's keep them separated because I can't possibly track all of them in one github issue.

This issue is about admin password no longer being accepted in UI and Unix (SSH, FTP). Your issue is about authenticating to the SMB server.

> RPi 3 running version 20170827 from Software Update (upgrade) Check button. Symptoms began after this update.
I see two symptoms:
> 1) Surveillance Username login at the browser GUI fails. Admin login still works fine here.
> 2) Mapped drives to motionEyeOS shared folders using smb:// from either OpenSUSE or Ubuntu fail.

I believe this has to do with IPv6. In any case I doubt it has anything to do with the authentication.

> open_socket_in(): socket() call failed: Address family not supported by protocol

ccrisan, 2017-09-13:
@jonwilliams84 that's a different issue that has been fixed upstream with https://github.com/ccrisan/motioneye/commit/7a8b5d78bee059f98bedd69d5cb3416a44d64c81. It'll be part of the next release.

> I think I am facing a similar issue with the surveillance user password. I have Basic auth turned on but whenever I try to login to the streaming URL (:8081) I get the Authentication Required box so [...]

ccrisan, 2017-09-13:
If I'm not mistaking, this issue (and not others that look similar) currently affects @clawwd, @jezmck, @jbradach, @magaa and @therocco. Guys, a few more questions for you:
1. Are you all on RPi Zero W?
2. I hate to ask but did you try clearing your browser's cache and see if the problem goes away? It may be using some old JS code that dealt differently with passwords.
3. Just to be clear, neither the UI nor SSH accepts your admin password anymore, right?

jezmck, 2017-09-13:
I was using a RPi Zero W, yes.
I've abandoned it for now though as the camera connector clip has broken.
I believe I must have tried clearing the cache, but this may not have cleared `localStorage` etc.

magaa, 2017-09-14:
1. Yes i have a Pi Zero W
2. Yes i clearing my Browser Cache, before and after the password change.
3. Yes no UI and SSH Admin access.
Its full reproducible. Every Time I change the Timezone or the Hostname after the reboot i have no Access. When I type my(or new) Admin password wenn I change Timezone or Hostname than it works.

clawwd, 2017-09-15:
I just configured a RPI1 and RPI2. Coincidence or not, I followed the same steps as @magaa in that I changed the admin password, timezone, and hostname at the same time. Everything is up and running with no problems.

OndraCerv, 2017-09-19:
Hello,

i have the same problem on RPI3.
Once I change any system settings, I can not sign in to restart.

jasaw, 2017-09-19:
I have difficulty reproducing this bug. Can anyone here please provide their hardware set up and step by step instructions on how to reproduce the problem?

Example:

Hardware Set Up:
- Raspberry Pi Zero W with class 10 micro SD card.
- Pi camera attached.
- Nothing else connected to USB ports.

Step by step instructions to reproduce the problem:
1. Boot up for the very first time from a freshly written image.
2. Login as admin.
3. Change admin and user password to _abcd1234_.
4. What other configuration changed before and after clicking _Apply_ ???
5. Did clicking the _Apply_ reboot the device ??? or did you reboot by clicking _Reboot_ button???
6. Login as admin with new password fails.

OndraCerv, 2017-09-19:
Hi, i will describe the procedure and configuration in the evening

rafi1975, 2017-09-19:
I'm experiencing the similar. I'm on RPI3 motionEyeOS 20170827. The box was set up ok and when trying to add to my hub I couldn't do that if there was a password set for 'admin'. It just doesn't allow me to add that Remote Motioneye camera. Once the password is blank I was able to add the camera. Not secure though.

jasaw, 2017-09-19:
@rafi1975 Your issue might be the same as #1137.

OndraCerv, 2017-09-21:
Hi, 

finally I came to the description.

Here is the step by step procedure:

1) SDcard formated from SDformatter 4.0 (SDcard is 2GB and 8GB class 10)
2) Used Win32 Disk Imager to burn a freshly downloaded copy of version 20170827
3) SDcard to RPi3, RPI connected by ethernet cable.
4) RPi3 turn on
5) First loggin  in web browser (Chrome Version 60.0.3112.113) via DHCP
6) Change admin and user password to abcd1234 and click to "Apply"
7) loggin
8) Click to "Reboot"
9) loggin (ok)
10) change Time zone (Europe/Prague) and Hostname (Kamery) click to "Apply"
11) question of motioneye (This will reboot the system. Continue?) click to "Yes"
12) Login (ok)
13) Configuration restore 
14) login with old password (ok)
15) Reboot
16) login with old password (ok)
17) Change admin and user password to abcd1234 and click to "Apply"
18) error message 
19) login with new password (ok)
20) cameras have stopped working (not a picture)
21) Reboot
22) login with new password (Ok)
23) cameras working (Ok)
24) change GPU Memory to 256 click to "apply"
25) question of motioneye (This will reboot the system. Continue?) click to "Yes"
26) login with new password (ko) -> Invalid credentials.

any change to settings from point 24 = Invalid credentials.

OndraCerv, 2017-09-21:
I tested the same behavior if I did not configuration restore and it seems ok.
I would see this problem with restore. 

excuse my English please :-)

OndraCerv, 2017-09-21:
so after an hour I tried to sign up and again ->  Invalid credentials.

jezmck, 2017-09-21:
I think the timezone might somehow be part of it.

ccrisan, 2017-09-21:
@OndraCerv you said something about `An error occurred. Refreshing is recommended` - something that other users haven't mentioned. This is a good lead. Could you please try to recover the motioneye.log file from your broken motionEyeOS sdcard? It's on the largest partition, at `/log/motioneye.log`. You'll need a Linux machine to read that partition.

OndraCerv, 2017-09-21:
@ccrisan ok, I will try to extract the log file.

OndraCerv, 2017-09-21:
@ccrisan 

do I have the log file where you want to send it?

jasaw, 2017-09-21:
@OndraCerv Try attaching the file here by dragging and dropping the file here.

ccrisan, 2017-09-22:
@OndraCerv I received your log by email and I found the following error:

    `KeyError: 'manual_snapshots'`

This is due to browser's cache. Can you please **clear your browser's cache** and try again all of the steps you described above?

OndraCerv, 2017-09-22:
ok. 
I'll check it out in a while with restore or not restore ?

ccrisan, 2017-09-22:
@OndraCerv no restore - your steps, from the beginning.

clawwd, 2017-09-22:
I will add that I have seen "Refreshing is recommended" as well while tweaking settings but I cant confirm if it happened during the initial setup of the cams when I run into the password problem. I can confirm I did not see this error on the last setup procedures I used which worked. Maybe one of the others that come here for answers can remember if they had gotten this message.

ccrisan, 2017-09-22:
@clawwd that message appears in the latest version because of the introduction of some new options (such as manual snapshots) that require a corresponding value coming from the UI. If the browser uses a cached version (older one), the server detects this problem and shows that error. This is why I insist on always clearing the cache.

Anyways, I'm not quite sure this is the root cause of our problem here.

OndraCerv, 2017-09-22:
@ccrisan after the browser cache was deleted, the password problem no longer occurred.
I have been testing for 12 hours any changes in the settings and the problem has not occurred.

Can the password contain special characters? I use an 18-character long password with special characters by default.

Thanks

ccrisan, 2017-09-22:
@OndraCerv that is good to hear. Yes, in theory you should be able to use complex passwords. I always try to make sure motionEye escapes any special characters in those passwords, as they are used in various places, as you can see.

jasaw, 2017-09-25:
@jezmck @magaa @therocco @jbradach Can you guys please **clear your browser cache** and try to reproduce the problem again?

sweh, 2017-09-26:
I'm facing the exact same problem. I changed TimeZone, too, after adding a password and rebooting the device.

Question: How can the browser cache be the cause for a no-longer working ssh login?

ccrisan, 2017-09-26:
@sweh in theory, if our assumptions are correct, your browser uses an older version of the javascript file that deals with the UI. When you make a change and click the Apply button, any new settings expected by the server won't be sent by the UI or in a different format.

The old UI would send the admin password with each apply and the OS would pick up that password (that was stored in clear text) and would set it as the root Unix password, at each boot.

Now the new UI will only send the admin password when it changes and the server immediately updates the root Unix password. The OS won't read it at boot and won't update anything anymore, as it's no longer stored in clear text.

Imagine now how old UI with new server can result in errors or simply, undesired root passwords :)

jasaw, 2017-09-27:
@ccrisan I think we should add a note to clear browser cache in the (pre)release.

ccrisan, 2017-09-27:
@jasaw I'm still not 100% convinced that the cache is the (only) root cause for this problem. Some of the users have **not** reported any errors and they claim they **did** clear the browser cache. I was hoping to get more feedback from them with regards to the browser cache and the reproducibility of the problem.

jezmck, 2017-09-27:
I guess it'd be best to add a version number to the script path, so that no old versions are used.

ccrisan, 2017-09-27:
@jezmck that makes sense. I'll see what I can do about that.

ccrisan, 2017-10-04:
Guys, if a cached version of the UI was running at the moment you applied your settings, then your actual password (as set by the mismatching server-side) could be one of the following variants:
 * `sha1("")`: `da39a3ee5e6b4b0d3255bfef95601890afd80709`
 * `sha1("null")`: `2be88ca4242c76e8253ac62474851065032d6833`
 * `sha1(<your admin password>)`

You can obtain the corresponding sha1 digest using the Linux command line (`echo -n "foo" | sha1sum`) or you could use http://www.sha1-online.com/.

Please let me know if one of the password variants above matches so we can be sure it's a cache problem.

crw, 2017-10-04:
I gave it a shot on both devices. I used the sha hashes suggested above as the input for the password field, none worked. Hard restarted both devices, tried again, still did not work. Tried in web and ssh.

crw, 2017-10-04:
Re-installing from scratch fixes / does not re-introduce the issue.

ccrisan, 2017-10-04:
@crw thansk for the update. When you say you tried the sha digests as the *input for the password field*, does that also include SSH or FTP or SMB or any kind of Unix login? motionEye login could not work due to the UI finally refreshing in the meantime.

crw, 2017-10-04:
I tried the original, un-hashed password as well as the three recommended hashes on both web and ssh (as root@). I did not try any other service.

crw, 2017-10-04:
A thought occurs: I could send you a copy of the image. I am not exactly sure how to make a perfect copy of the disk image, but I would be happy to do so.

ccrisan, 2017-10-04:
If you still have the broken SD card, it would be very helpful if you could send it over to me. You mentioned you reinstalled from scratch so I assumed everything was lost.

If you're a Windows person, you could use Win32DiskImager to **read** the SD card (just like you would do to write it).

If you're a Linux or macOS person, you could use dd (e.g. `dd if=/dev/[your card device] of=broken.img bs=100M`).

Then you should compress the img file (using whatever compressor is more convenient for you) and drop me an email at ccrisan at gmail dot com so that I can reply with details of my FTP server, where you can upload it.

[15 more replies]

closed 2017-10-06 by ccrisan