motioneye-project/motioneyeos #1119: Firmware upgrade supports XZ compressed image

jasaw, 2017-09-06:
Current GZ disk images are around 70MB each for Raspberry Pis. When upgrading firmware, most of the time is spent downloading the disk image.

Using XZ compression, the same image can be compressed down to 45MB.
Example compression command:
`XZ_OPT=-9 tar cJf motioneyeos-raspberrypi.tar.xz motioneyeos-raspberrypi.img`

I don't know how much memory is required and how long it takes to extract an XZ file, but it may be a feasible option. If memory and CPU does become a concern, motionEye can always kill off motion software before starting the extract and upgrade process (not sure if it's already doing this).

Thoughts?

ccrisan, 2017-09-06:
This is actually a very good idea. This needs to be tested and I think the firmware upgrade mechanism can be implemented to be backwards compatible.

One more little thing: the firmware upgrade mechanism, as well as many other common OS parts, have been moved to [thingOS](https://github.com/ccrisan/thingos/). This issue actually belongs to thingOS. Can we move issues across projects on github?

jasaw, 2017-09-06:
I've created this issue as https://github.com/ccrisan/thingos/issues/4

jasaw, 2018-03-01:
PR merged into thingos.

closed 2018-03-01 by jasaw